ARMmbed/mbed-os #14855: Investigate which tool (mbedgt or htrun) we use for CTest

LDong-Arm, 2021-07-01:
### Description of defect

Mbed CLI 1 generates `test_spec.json` (listing all tests and their image locations) and passes it to `mbedgt`, which performs
* detection of connected targets
* orchestration of tests
* invoking of `mbedhtrun` to run each test

With Mbed CLI 2, orchestration of tests will be done by CTest. So we need to investigate whether to use `mbedgt` or `mbedhtrun`:
* `mbedgt` has advantage of target detection, but requires test_spec.json which is redundant with CTest which runs tests one-by-one
* `mbedhtrun` is good for running one test, but lacks detection of existing target

We need to investigate with one to use, and if any changes in either tool is needed.

#### Target(s) affected by this defect ?

N/A

#### Toolchain(s) (name and version) displaying this defect ?

N/A

#### What version of Mbed-os are you using (tag or sha) ?

N/A

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

N/A

#### How is this defect reproduced ? 

N/A

LDong-Arm, 2021-07-05:
@ARMmbed/mbed-os-core 

Personally I think htrun is preferable for integration with CTest, because htrun supports all the command line options needed for running a single test (image, serial port, host scripts path, sync packets, …). It can do flashing + testing in one go, as long as an image path is provided. And mbedgt is simply translates test_spec.json into htrun commands which does the testing, but we will use CTest to drive the high-level flow of all tests anyway.

I’ve never had any issues syncing and running tests with htrun directly.

As for serial port and mount point detection, mbedgt uses legacy CLI 1 code which we are moving away from. Instead, it should be quite easy to use the new mbed-tools for detection:
```
mbedtools detect --format json
```
Then in CMake we can extract port and mount point from JSON output and pass them to htrun. This detection is needed once only for all tests.

Patater, 2021-07-05:
This sounds like a good approach to me, too. The "libxo" style integration interface (json) should be easier to test and guarantee than the legacy Python library interface of CLI 1.

LDong-Arm, 2021-07-05:
> This sounds like a good approach to me, too. The "libxo" style integration interface (json) should be easier to test and guarantee than the legacy Python library interface of CLI 1.

In this way I believe the tools already provide what we need in order to start adding CTest support for all tests in Mbed OS. As others also seem okay with this approach, the investigation is done and I'm closing this issue.

closed 2021-07-05 by LDong-Arm