ARMmbed/mbed-os #14860: Incompatibility between master and docker image

pan-, 2021-07-02:
### Description of defect

Change of requirements in `mbed-os/requirements.txt` has broken use of latest mbed-os with he latest mbed-os docker image. 

To test with the latest master we run the latest docker image then install requirements from the mbed-os version used. 
https://github.com/ARMmbed/mbed-os/pull/14780 has introduced new requirements to `urllib3` which are not compatible with the requirements required by mbed-ls present in the latest mbed-os docker image publish. 

As a result a call to mbed-ls in the docker image fail after the latest mbed-os requirements have been installed. 

#### Target(s) affected by this defect ?

All

#### Toolchain(s) (name and version) displaying this defect ?

All

#### What version of Mbed-os are you using (tag or sha) ?
Latest master

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Latest in mbed-os docker image.

#### How is this defect reproduced ? 

```shell
docker run -it mbedos/mbed-os-env:latest /bin/bash
wget https://raw.githubusercontent.com/ARMmbed/mbed-os/master/requirements.txt
pip3 install -r requirements.txt
mbedls
```

> Traceback (most recent call last):
  File "/usr/local/bin/mbedls", line 8, in <module>
    sys.exit(mbedls_main())
  File "/usr/local/lib/python3.8/dist-packages/mbed_lstools/main.py", line 185, in mbedls_main
    logger.debug("mbed-ls ver. %s", get_version())
  File "/usr/local/lib/python3.8/dist-packages/mbed_lstools/main.py", line 41, in get_version
    return pkg_resources.require("mbed-ls")[0].version
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 901, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 792, in resolve
    raise VersionConflict(dist, req).with_context(dependent_req)
pkg_resources.ContextualVersionConflict: (urllib3 1.26.6 (/usr/local/lib/python3.8/dist-packages), Requirement.parse('urllib3<1.25,>=1.21.1'), {'requests'})

LDong-Arm, 2021-07-02:
> pkg_resources.ContextualVersionConflict: (urllib3 1.26.6 (/usr/local/lib/python3.8/dist-packages), Requirement.parse('urllib3<1.25,>=1.21.1'), {'requests'})

Maybe we can bump the version of urllib3 used by mbed-ls and make a release?

Patater, 2021-07-02:
Latest requests library has `'urllib3>=1.21.1,<1.27',` (https://github.com/psf/requests/blob/master/setup.py#L46), so probably someone is trying to use an old requests version. It's not mbed-os-tools, as we are OK with any requests library `requests>=2.0,<3.0`. I'll check mbed-os next.

Patater, 2021-07-02:
The workflow given to reproduce the issue seems questionable to me. Shouldn't the Docker image already have the mbed-os python requirements installed? Why the need to run `pip3 install -r requirements.txt`?

@saheerb What's the expectation here with the Docker image? Should it be good to go out of the box with all the dependencies required (including python dependencies) for the latest mbed-os already installed and good to go?

pan-, 2021-07-02:
> The workflow given to reproduce the issue seems questionable to me. Shouldn't the Docker image already have the mbed-os python requirements installed? Why the need to run pip3 install -r requirements.txt?

It indeed is questionable but until we have a released mbed-os docker image synced with master, that's one way to add extra requirements from master.

Patater, 2021-07-02:
I thought this was released synced with master. If not, alright, the workflow makes more sense now.

What happens if you try `pip install --upgrade-strategy=eager -r requirements.txt`?

pan-, 2021-07-02:
It returns exactly the same error. The issue is that requirements of mbed-ls are incompatible with `mbed-ls`.

- mbed-ls:  urllib3<1.25,>=1.21.1
- mbed-os: urllib==1.26.6

LDong-Arm, 2021-07-02:
> pkg_resources.ContextualVersionConflict: (urllib3 1.26.6 (/usr/local/lib/python3.8/dist-packages), Requirement.parse('urllib3<1.25,>=1.21.1'), {'requests'})

The error above does say the version requirements are set by "requests" (not mbed-ls), and as @Patater suggested `--upgrade-strategy=eager` should select newer versions (while keeping compatibility?). So, why do we still end up with an older version of "requests"?

Is there a chance some other modules listed in `requirements.txt` does require an older version of "requests", so pip resolves to an older one? I'm not an export in pip though...

saheerb, 2021-07-02:
> The workflow given to reproduce the issue seems questionable to me. Shouldn't the Docker image already have the mbed-os python requirements installed? Why the need to run `pip3 install -r [...]

The work to update  mbed-os-env image and keep in sync with master is in progress. Current version of mbed-os-env:latest is about 25 days old. 

I get the error mentioned in this issue when I try to rebuild the docker image.

Patater, 2021-07-02:
> It returns exactly the same error. The issue is that requirements of mbed-ls are incompatible with `mbed-ls`.
> 
>     * mbed-ls:  urllib3<1.25,>=1.21.1
> 
>     * mbed-os: urllib==1.26.6

That mbed-os requirement doesn't look right to me. https://github.com/ARMmbed/mbed-os/blob/master/requirements.txt#L2 shows it as ">=1.26.5", not "==1.26.6".

The mbed-ls requirement also doesn't look right to me. mbed-ls allows (even 25 days ago) newer `requests` than the version of `requests` that required urllib3<1.25,>=1.21.1.

We'll need to debug to see where the actual issue is coming from.

saheerb, 2021-07-02:
The version conflicts we see here could be because we haven't built docker image for 25 days. Ideally, rebuilding the image should work. I am currently working on auto rebuilding docker image which should hopefully land soon.

I face another conflict when rebuilding the image though...

`#13 1.094 pkg_resources.ContextualVersionConflict: (prettytable 0.7.2 (/usr/local/lib/python3.8/dist-packages), Requirement.parse('prettytable<3.0,>=2.0; python_version >= "3.6"'), {'mbed-os-tools'})`

This seems to be due to the icetea dependencies:

https://github.com/ARMmbed/mbed-os/blob/master/requirements.txt#L21
https://github.com/ARMmbed/icetea/blob/master/requirements.txt#L1
https://github.com/ARMmbed/mbed-os-tools/blob/master/requirements.txt#L6

Patater, 2021-07-06:
Raised https://github.com/ARMmbed/icetea/pull/115 to align icetea dependency on prettytable with other mbed libraries.

saheerb, 2021-07-06:
> Raised [ARMmbed/icetea#115](https://github.com/ARMmbed/icetea/pull/115) to align icetea dependency on prettytable with other mbed libraries.

I don't think icetea is a dependency anymore, may be we can remove it? @jamesbeyond ?

Patater, 2021-07-07:
I've raised https://github.com/ARMmbed/mbed-os/pull/14880 to Mbed OS instead, as icetea is deprecated and won't have new releases made. This will remove icetea as a default dependency of Mbed OS, requiring you to install it separately (probably to a different virtualenv) if you want it.

jamesbeyond, 2021-07-07:
> > Raised [ARMmbed/icetea#115](https://github.com/ARMmbed/icetea/pull/115) to align icetea dependency on prettytable with other mbed libraries.
> 
> We will need to update the[ ice-tea dependency version ](https://github.com/ARMmbed/mbed-os/blob/master/requirements.txt#L21)in mbed-os as well.
> 
> I don't think icetea is a dependency anymore, we can remove it? @jamesbeyond ?

Yeah go ahead and remove icetea, it is not required anymore