ARMmbed/mbed-os #14863: Refactor connectivity unit tests to use the minimum amount of C++ files

rajkan01, 2021-07-02:
### Description of defect
Refactor connectivity unit tests to use the minimum amount of C++ files

#### Target(s) affected by this defect ?
N/A

#### Toolchain(s) (name and version) displaying this defect ?
N/A

#### What version of Mbed-os are you using (tag or sha) ?

Latest

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
N/A

#### How is this defect reproduced ? 
N/A

mbedmain, 2021-07-02:
@rajkan01 thank you for raising this issue.Please take a look at the following comments:

Could you add some more detail to the description? A good description should be at least 25 words.

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

paul-szczepanek-arm, 2021-07-05:
What is the issue?