ARMmbed/mbed-os #14866: Incorrect gcc_arm linker script name for stm32f407xg

byq77, 2021-07-02:
### Description of defect

So I have a custom board that incorporates the `stm32f407xG` mcu. I was able to successfully configure a custom target using Mbed Os 6.12 and CLI2. I followed [this](https://github.com/ARMmbed/mbed-os-5-docs/blob/development/docs/porting/custom-target-porting.md) guide. However, I found that the linker script name for this model is incorrect for some reason. It is in uppercase, but the `CMakeLists.txt` in the `mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F407xG/` directory reference to the name in lowercase. It looks like the lowercase is also the naming convention you use. Converting  `STM32F407XG.ld` to `stm32f407xg.ld` fixed the issue.

#### Target(s) affected by this defect ?

- `custom_target.json` -> `stm32f407xG`

#### Toolchain(s) (name and version) displaying this defect ?

- `GCC_ARM`

#### What version of Mbed-os are you using (tag or sha) ?

- `mbed-os-6.12.0`

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

- `mbed CLI2`

#### How is this defect reproduced ?

mbedmain, 2021-07-02:
@byq77 thank you for raising this issue.Please take a look at the following comments:

How can we reproduce your issue?

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

jeromecoutant, 2021-07-07:
> It looks like the lowercase is also the naming convention you use. Converting `STM32F407XG.ld` to `stm32f407xg.ld` fixed the issue.

Hi

I confirm. Please raise a pull request, we will be happy to integrate it asasp.

Thx

byq77, 2021-07-08:
https://github.com/ARMmbed/mbed-os/pull/14898

closed 2021-07-09 by 0xc0170