ARMmbed/mbed-os #14867: Artemis Thing Plus: printf("%s") strange characters in output

idea--list, 2021-07-03:
### Description of defect

Artemis Thing Plus seems to output some strange and unexpected characters with `printf(%s);` but the output gets fixed if code also contains `std::string`

#### Target(s) affected by this defect ?
Artemis Thing Plus (probably all targets with Ambiq Apollo3 MCU)

#### Toolchain(s) (name and version) displaying this defect ?
GCC_ARM 8 2018-q4-major

#### What version of Mbed-os are you using (tag or sha) ?

Mbed OS 6.12

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Mbed Studio V1.4.1

#### How is this defect reproduced ? 

Compile&run this code, then uncomment the commented lines and notice the difference:

```
#include "mbed.h"
#include <cstdint>
#include <cstdio>
#include <string>

// main() runs in its own thread in the OS
struct astruct {
    char name[3];
    //std::string text;
    uint8_t ID;
};

int main()
{
    astruct struct1;
    strcpy(struct1.name, "ABC");
    //struct1.text = "xyZ";
    struct1.ID = 1;

    //printf("name: %s\ntext: %s\nID: %d\n", struct1.name, struct1.text.c_str(), struct1.ID);
    printf("name: %s\nID: %d\n", struct1.name, struct1.ID);

    while (true) {

    }
    return 0;
}
```

Initial output in serial monitor:
```
name: ABC`Q
ID: 1
```

mbedmain, 2021-07-03:
@idea--list thank you for raising this issue.Please take a look at the following comments:

Could you add some more detail to the description? A good description should be at least 25 words.

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

16L-YT, 2021-07-04:
I tried the example my side using a costume made Apollow3 based board, and the `printf` output was good: 

[image]

Maybe the corrupted data due to the usb-serial converter or the terminal program. I use FT2232H chip as a converter powered with 3.3V and my Apollo3 with 1.8V and TeraTerm as a console.

idea--list, 2021-07-04:
@16L-YT Thanks for trying!

closed 2021-07-04 by idea--list