ARMmbed/mbed-os #14869: `ChainableGapEventHandler` events are not called (namly `onConnectionComplete` and `onDisconnectionComplete` ).

16L-YT, 2021-07-04:
### Description of defect
Using [ LinkLoss BLE service](https://github.com/ARMmbed/mbed-os-experimental-ble-services/tree/main/services/LinkLoss), `onConnectionComplete` and `onDisconnectionComplete` callbacks seem not to be called, as I added a tracing logs inside each callback. I also added the `onConnectionComplete` and `onDisconnectionComplete` callbacks to another local BLE service using the same way of LinkLoss class of overriding the callbacks of `ChainableGapEventHandler` and registration of eventhandler. 
 `_chainable_gap_event_handler.addEventHandler(this);`

I found another issue of `onDisconnectionComplete` callback but for `ble::Gap::EventHandler` class, that if I made the disconnection or link lost while notifications are being sent, the callback will not be called too. If I made the disconnection or link loss with no ongoing activity (like sending notifications with high frequency ~5Hz), then the `onDisconnectionComplete` is called correctly, otherwise it will not be called. I tried to go back to the initial place of calling this callback and did not find an explanation for this behavior. 

#### Target(s) affected by this defect ?
Issue does not seem to be target specific.

#### Toolchain(s) (name and version) displaying this defect ?
gcc version 10.2.1 20201103 (release) (xPack GNU Arm Embedded GCC, 64-bit)

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os-6.9.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Mbed Studio 1.4.1

#### How is this defect reproduced ? 
* To reproduce `ChainableGapEventHandler` issue, add any tracing message to  `onConnectionComplete` and `onDisconnectionComplete` inside LinkLoss class implementation. 
* To reproduce `onDisconnectionComplete` issue from `ble::Gap::EventHandler` enable tracing and try to disconnect during high frequency of sending notifications.

paul-szczepanek-arm, 2021-07-05:
Cannot reproduce. Please post your code.

16L-YT, 2021-07-12:
I can't share the whole project. I will try to reproduce with on of the ready examples and report back. 

BTW, the issue appears only with tracing enabled.

paul-szczepanek-arm, 2021-07-14:
Right now I cannot reproduce this bug. Please reopen this issue when you have some code I can use to repro your problem.

closed 2021-07-14 by paul-szczepanek-arm