ARMmbed/mbed-os #14885: how to used arm_pack_manager

sitieqiang, 2021-07-07:
I write a pdsc file  for my device,how to generate  index.json？

mbedmain, 2021-07-07:
@sitieqiang thank you for raising this issue.Please take a look at the following comments:

Could you add some more detail to the description? A good description should be at least 25 words.
What target(s) are you using?
What toolchain(s) are you using?
What Mbed OS version are you using?
It would help if you could also specify the versions of any tools you are using?
 How can we reproduce your issue?

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

sitieqiang, 2021-07-07:
i check out code at today

closed 2021-07-08 by sitieqiang