ARMmbed/mbed-os #14907: Set up CI testing for greentea using CTest.

rwalton-arm, 2021-07-09:
### Description of defect
Enable CI testing for CTest driven greentea tests to catch failures and regressions.

#### Target(s) affected by this defect ?
N/A

#### Toolchain(s) (name and version) displaying this defect ?
N/A

#### What version of Mbed-os are you using (tag or sha) ?

N/A

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
N/A

#### How is this defect reproduced ? 
N/A

mbedmain, 2021-07-09:
@rwalton-arm thank you for raising this issue.Please take a look at the following comments:

Could you add some more detail to the description? A good description should be at least 25 words.

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

rwalton-arm, 2021-08-10:
@saheerb Has added CI scripts to run the greentea tests using CTest.

closed 2021-08-10 by rwalton-arm