ARMmbed/mbed-os #14916: mbed_application_start() not working on NUCLEO_F070RB

ilmanatz, 2021-07-13:
### Description of defect
When writing a simple bootloader, calling the function mbed_application_start() results in a compilation error _use of undeclared identifier 'mbed_start_application'_; this appears to be the case only with this platform, at least that I am aware of. Compiling the same code for the nRF52_DK (only other mbed board that I have available) works fine.

#### Target(s) affected by this defect ?
NUCLEO_F070RB

#### Toolchain(s) (name and version) displaying this defect ?
ARMC6

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os-6.8.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed Studio 1.4.1
ARM Compiler 6.15

#### How is this defect reproduced ? 
Simplest main.cpp file possibile, like:

```c
#include "mbed.h"
int main(){
     mbed_application_start(POST_APPLICATION_ADDR);
}
```

For reference, I used the following mbed_app.json file. It doesn't seem related to the issue anyway, but it's here if it ends up being the case.
```json
{
    "target_overrides": { 
        "*": {
            "target.clock_source": "USE_PLL_HSI",
            "target.features_add" : [ "STORAGE" ],
            "target.components_add": [ "SPIF", "SD" ],
            "sd.SPI_MOSI" : "D11",
            "sd.SPI_MISO" : "D12",
            "sd.SPI_CLK" : "D13",
            "sd.SPI_CS" : "D10",
            "spif-driver.SPI_MOSI" : "D11",
            "spif-driver.SPI_MISO" : "D12",
            "spif-driver.SPI_CLK": "D13",
            "spif-driver.SPI_CS" : "D9",
            "platform.stdio-baud-rate": 9600,
            "platform.default-serial-baud-rate": 9600,
            "platform.stdio-flush-at-exit": false,
            "platform.stdio-buffered-serial": false,
            "rtos.main-thread-stack-size": 768,
            "rtos.thread-stack-size" : 512,
            "rtos.timer-thread-stack-size" : 512,
            "rtos.idle-thread-stack-size": 512,
            "drivers.uart-serial-rxbuf-size": 8,
            "drivers.uart-serial-txbuf-size": 8,
            "events.shared-dispatch-from-application" : true
        },
        "NRF52_DK":{
            "target.restrict_size": "0x12000"
        },
        "NUCLEO_F070RB":{
            "target.restrict_size": "0x10000"
        }
    },
    "macros":[
        "NDEBUG=1"
    ]
}
```

0xc0170, 2021-07-14:
Does `NUCLEO_F070RB` support bootloader? I don't see it enabled in targets.json for this target.

jeromecoutant, 2021-07-15:
Hi

> Does `NUCLEO_F070RB` support bootloader? I don't see it enabled in targets.json for this target.

Yes, bootloader is enabled at "MCU_STM32" level.

But it seems that SystemInit is empty...
https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32F0/STM32Cube_FW/system_stm32f0xx.c#L128

Could you try to add 
````
#include "nvic_addr.h"
SCB->VTOR = NVIC_FLASH_VECTOR_ADDRESS;
````
like for STM32F4 for ex:
https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32F4/STM32Cube_FW/system_stm32f4xx.c#L169

ilmanatz, 2021-07-15:
Hi, I tried to introduce the change proposed
```
#include "nvic_addr.h"
SCB->VTOR = NVIC_FLASH_VECTOR_ADDRESS;
```
in the file you suggested, but the stack still does not compile, raising the same error
```
.\main.cpp:5:9: error: use of undeclared identifier 'mbed_start_application'
mbed_start_application(POST_APPLICATION_ADDR);
```

jeromecoutant, 2021-07-15:
In fact issue seems that STM32F0 is cortex M0, which doesn't support mbed_start_application
(NRF52 is M4)

https://github.com/ARMmbed/mbed-os/blob/master/platform/include/platform/mbed_application.h#L23

ilmanatz, 2021-07-15:
I was 100% sure that the F070RB was in fact an M0+ core, and not a plain M0.
Sorry for wasting everybody's time. Apparently it's just not supported. I will change development platform since bootloader is considered mission critical.
Thanks for your help and sorry again

closed 2021-07-19 by 0xc0170