ARMmbed/mbed-os #14918: Quectel BC95-G AT-command doesn't require quotation marks

alcheagle, 2021-07-14:
### Description of defect
The cellular module BC-95-G is returning code `50`, meaning 'incorrect parameters '. By inspecting the AT debug mode I noticed the commands had quotation marks surrounding the hex string, from Quectel AT datasheet I saw they are not requiring them.

If I comment the following lines the device started working.
https://github.com/ARMmbed/mbed-os/blob/cecc47b4a53951527dd3f670465c8566396ad101/connectivity/cellular/source/framework/device/ATHandler.cpp#L1561
https://github.com/ARMmbed/mbed-os/blob/cecc47b4a53951527dd3f670465c8566396ad101/connectivity/cellular/source/framework/device/ATHandler.cpp#L1568

Logs of the non working version:
```
[00003788ms][DBG ][main]: Connection Established.
[00003794ms][INFO][CELL]: Socket 0 open
[00003799ms][INFO][CELL]: AT TX (15): AT+NSOCR=DGRAM,
[00003805ms][INFO][CELL]: AT TX ( 2): 17
[00003809ms][INFO][CELL]: AT TX ( 1): ,
[00003814ms][INFO][CELL]: AT TX ( 5): 51791
[00003819ms][INFO][CELL]: AT TX ( 1): ,
[00003824ms][INFO][CELL]: AT TX ( 1): 1
[00003829ms][INFO][CELL]: AT TX ( 1): ,
[00003834ms][INFO][CELL]: AT TX ( 1): "
[00003838ms][INFO][CELL]: AT TX ( 7): AF_INET
[00003844ms][INFO][CELL]: AT TX ( 1): "
[00003848ms][INFO][CELL]: AT TX ( 1): <cr>
[00003853ms][INFO][CELL]: AT RX ( 2): <cr><ln>
[00003859ms][INFO][CELL]: AT RX ( 6): 1<cr><ln><cr><ln>O
[00003865ms][INFO][CELL]: AT RX ( 3): K<cr><ln>
[00003871ms][INFO][CELL]: Socket create id: 1
[00003876ms][INFO][CELL]: AT TX ( 9): AT+NSOST=
[00003881ms][INFO][CELL]: AT TX ( 1): 1
[00003886ms][INFO][CELL]: AT TX ( 1): ,
[00003891ms][INFO][CELL]: AT TX (15): 185.215.195.137
[00003897ms][INFO][CELL]: AT TX ( 1): ,
[00003902ms][INFO][CELL]: AT TX ( 1): 7
[00003906ms][INFO][CELL]: AT TX ( 1): ,
[00003911ms][INFO][CELL]: AT TX ( 1): 4
[00003916ms][INFO][CELL]: AT TX ( 1): ,
[00003921ms][INFO][CELL]: AT TX ( 1): "
[00003925ms][INFO][CELL]: AT TX ( 2): 54
[00003930ms][INFO][CELL]: AT TX ( 2): 45
[00003935ms][INFO][CELL]: AT TX ( 2): 53
[00003940ms][INFO][CELL]: AT TX ( 2): 54
[00003945ms][INFO][CELL]: AT TX ( 1): "
[00003950ms][INFO][CELL]: AT TX ( 1): <cr>
[00003955ms][INFO][CELL]: AT RX ( 2): <cr><ln>
[00003960ms][INFO][CELL]: AT RX ( 5): +CME 
[00003965ms][INFO][CELL]: AT RX ( 5): ERROR
[00003970ms][INFO][CELL]: AT RX ( 5): : 50<cr>
[00003976ms][INFO][CELL]: AT RX ( 1): <ln>
[00003981ms][WARN][CELL]: AT error code 50
[00003986ms][DBG ][CELL]: AT error -3012
[00003991ms][ERR ][CELL]: Socket 0 sendto 185.215.195.137 error -3012
[00003998ms][DBG ][main]: UDPSocket.send() fails, code: -3012
[00004005ms][DBG ][main]: Sending and received data failed.
[00004011ms][INFO][CELL]: CellularContext disconnect()
[00004017ms][INFO][CELL]: AT TX ( 9): AT+CGACT?
[00004023ms][INFO][CELL]: AT TX ( 1): <cr>
[00004031ms][INFO][CELL]: AT RX ( 1): <cr>
[00004033ms][INFO][CELL]: AT RX ( 2): <ln>+
[00004038ms][INFO][CELL]: AT RX ( 5): CGACT
[00004043ms][INFO][CELL]: AT RX ( 5): :0,1<cr>
[00004048ms][INFO][CELL]: AT RX ( 4): <ln><cr><ln>+
[00004054ms][DBG ][CELL]: Found active context
[00004060ms][INFO][CELL]: AT RX (12): CGACT:1,0<cr><ln><cr>
[00004066ms][INFO][CELL]: AT RX ( 5): <ln>OK<cr><ln>
[00004072ms][DBG ][CELL]: callback: 0, ptr: 2
[00004077ms][INFO][CELL]: cb: CellularContext disconnected
[00004084ms][INFO][CELL]: CellularContext disconnected
[00004090ms][DBG ][main]: Success. Exiting
[00004095ms][INFO][CELL]: AT TX ( 9): AT+NSOCL=
[00004100ms][INFO][CELL]: AT TX ( 1): 1
[00004105ms][INFO][CELL]: AT TX ( 1): <cr>
[00004110ms][INFO][CELL]: AT RX ( 2): <cr><ln>
[00004115ms][INFO][CELL]: AT RX ( 4): OK<cr><ln>
[00004121ms][INFO][CELL]: Close socket: 1 error: 0
[00004127ms][INFO][CELL]: Socket 0 closed
```

Logs of the working version with the quoted lines commented: 
```
[00003798ms][DBG ][main]: Connection Established.
[00003804ms][INFO][CELL]: Socket 0 open
[00003809ms][INFO][CELL]: AT TX (15): AT+NSOCR=DGRAM,
[00003815ms][INFO][CELL]: AT TX ( 2): 17
[00003819ms][INFO][CELL]: AT TX ( 1): ,
[00003824ms][INFO][CELL]: AT TX ( 5): 53042
[00003829ms][INFO][CELL]: AT TX ( 1): ,
[00003834ms][INFO][CELL]: AT TX ( 1): 1
[00003839ms][INFO][CELL]: AT TX ( 1): ,
[00003844ms][INFO][CELL]: AT TX ( 1): "
[00003848ms][INFO][CELL]: AT TX ( 7): AF_INET
[00003854ms][INFO][CELL]: AT TX ( 1): "
[00003858ms][INFO][CELL]: AT TX ( 1): <cr>
[00003863ms][INFO][CELL]: AT RX ( 2): <cr><ln>
[00003869ms][INFO][CELL]: AT RX ( 6): 1<cr><ln><cr><ln>O
[00003875ms][INFO][CELL]: AT RX ( 3): K<cr><ln>
[00003881ms][INFO][CELL]: Socket create id: 1
[00003886ms][INFO][CELL]: AT TX ( 9): AT+NSOST=
[00003891ms][INFO][CELL]: AT TX ( 1): 1
[00003896ms][INFO][CELL]: AT TX ( 1): ,
[00003901ms][INFO][CELL]: AT TX (15): 185.215.195.137
[00003907ms][INFO][CELL]: AT TX ( 1): ,
[00003912ms][INFO][CELL]: AT TX ( 1): 7
[00003916ms][INFO][CELL]: AT TX ( 1): ,
[00003921ms][INFO][CELL]: AT TX ( 1): 4
[00003926ms][INFO][CELL]: AT TX ( 1): ,
[00003931ms][INFO][CELL]: AT TX ( 2): 54
[00003936ms][INFO][CELL]: AT TX ( 2): 45
[00003940ms][INFO][CELL]: AT TX ( 2): 53
[00003945ms][INFO][CELL]: AT TX ( 2): 54
[00003950ms][INFO][CELL]: AT TX ( 1): <cr>
[00003955ms][INFO][CELL]: AT RX ( 2): <cr><ln>
[00003961ms][INFO][CELL]: AT RX ( 6): 1,4<cr><ln><cr>
[00003967ms][INFO][CELL]: AT RX ( 5): <ln>OK<cr><ln>
[00003972ms][INFO][CELL]: Socket 0 sent 4 bytes to 185.215.195.137 port 7
[00003980ms][DBG ][main]: UDP: Sent 4 Bytes to 185.215.195.137
[00003987ms][INFO][CELL]: AT TX ( 9): AT+NSORF=
[00003992ms][INFO][CELL]: AT TX ( 1): 1
[00003997ms][INFO][CELL]: AT TX ( 1): ,
[00004002ms][INFO][CELL]: AT TX ( 1): 4
[00004007ms][INFO][CELL]: AT TX ( 1): <cr>
[00004012ms][INFO][CELL]: AT RX ( 3): <cr><ln>O
[00004017ms][INFO][CELL]: AT RX ( 3): K<cr><ln>
[00004023ms][INFO][CELL]: AT TX ( 9): AT+NSORF=
[00004028ms][INFO][CELL]: AT TX ( 1): 1
[00004033ms][INFO][CELL]: AT TX ( 1): ,
[00004038ms][INFO][CELL]: AT TX ( 1): 4
[00004042ms][INFO][CELL]: AT TX ( 1): <cr>
[00004047ms][INFO][CELL]: AT RX ( 3): <cr><ln>O
[00004053ms][INFO][CELL]: AT RX ( 3): K<cr><ln>
[00005066ms][DBG ][CELL]: AT OoB readable 1, len 0
[00005067ms][INFO][CELL]: AT RX ( 1): <cr>
[00005068ms][INFO][CELL]: AT RX ( 1): <ln>
[00005069ms][INFO][CELL]: AT RX ( 1): +
[00005070ms][INFO][CELL]: AT RX ( 1): N
[00005071ms][INFO][CELL]: AT RX ( 1): S
[00005075ms][INFO][CELL]: AT RX ( 4): ONMI
[00005080ms][INFO][CELL]: AT RX ( 4): :1,4
[00005085ms][INFO][CELL]: AT RX ( 2): <cr><ln>
[00005091ms][DBG ][CELL]: AT OoB done
[00005095ms][INFO][CELL]: AT TX ( 9): AT+NSORF=
[00005101ms][INFO][CELL]: AT TX ( 1): 1
[00005106ms][INFO][CELL]: AT TX ( 1): ,
[00005110ms][INFO][CELL]: AT TX ( 1): 4
[00005115ms][INFO][CELL]: AT TX ( 1): <cr>
[00005120ms][INFO][CELL]: AT RX ( 3): <cr><ln>1
[00005126ms][INFO][CELL]: AT RX ( 6): ,185.2
[00005131ms][INFO][CELL]: AT RX ( 4): 15.1
[00005136ms][INFO][CELL]: AT RX ( 5): 95.13
[00005141ms][INFO][CELL]: AT RX ( 5): 7,7,4
[00005146ms][INFO][CELL]: AT RX ( 5): ,5445
[00005151ms][INFO][CELL]: AT RX ( 5): 5354,
[00005156ms][INFO][CELL]: AT RX ( 4): 0<cr><ln><cr>
[00005162ms][INFO][CELL]: AT RX ( 5): <ln>OK<cr><ln>
[00005168ms][INFO][CELL]: Socket 0 recv 4 bytes from 185.215.195.137 port 7
[00005176ms][DBG ][main]: received buffer: "TEST"
[00005181ms][DBG ][main]: Received from echo server 4 Bytes
```

I am using a slightly modified version of https://github.com/ARMmbed/mbed-os-example-cellular in order to include more debug info.

#### Target(s) affected by this defect ?

#### Toolchain(s) (name and version) displaying this defect ?

#### What version of Mbed-os are you using (tag or sha) ?
https://github.com/ARMmbed/mbed-os/#cecc47b4a53951527dd3f670465c8566396ad101

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ?

mbedmain, 2021-07-14:
@alcheagle thank you for raising this issue.Please take a look at the following comments:

What target(s) are you using?
What toolchain(s) are you using?
It would help if you could also specify the versions of any tools you are using?
 How can we reproduce your issue?

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

paul-szczepanek-arm, 2021-07-14:
Looks like you're right. We can't break the command for other modules though. How about we fix it driver level?

https://github.com/ARMmbed/mbed-os/pull/14922

I don't know what other modules are supported. I checked bg96 spec and it also doesn't need the quotes but people might be using the at handler for their devices so we can only fix the devices at the driver level that we already support.

@jarvte do you know anything about this discrepancy between our code and the spec? Is there variability in devices, should this be a configuration option?

0xc0170, 2021-07-20:
#14922 was merged, shall this be closed (besides some questions above, discussion can still be ongoing) ?

alcheagle, 2021-07-20:
I I think the issue can be closed since It had been resolved by the PR you are mentioning

closed 2021-07-20 by alcheagle