ARMmbed/mbed-os #14923: mac filter can not add address again when remove the last address

mcuxmx, 2021-07-15:
### Description of defect
When mac_ filter_ delete_ The long() function removes the last eui64 address. Cannot take effect when adding the eui64 address again.
But if the deletion is not the last content of the list, it will not be affected. It may be a problem of ns_list_remove() function.

like:
uint8_t eui64_1[8] = {0,0,0,0,0,0,0,1};
uint8_t eui64_2[8] = {0,0,0,0,0,0,0,2};
uint8_t eui64_3[8] = {0,0,0,0,0,0,0,3};

mac_filter_start( 1, MAC_FILTER_BLOCKED );
mac_filter_add_long( 1, eui64_1,  MAC_FILTER_ALLOWED );
mac_filter_add_long( 1, eui64_2,  MAC_FILTER_ALLOWED );
mac_filter_delete_long( 1, eui64_2 ); 
mac_filter_add_long( 1, eui64_3,  MAC_FILTER_ALLOWED ); // eui64_ 3 will not be added to the list

#### Target(s) affected by this defect ?
BR: DISCO_F769NI+S2LP radio
NODE: NUCLEO_F429ZI+S2LP radio

#### Toolchain(s) (name and version) displaying this defect ?
mbed cli 1.10.4

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-5.15.4

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
any

#### How is this defect reproduced ? 
alyways

mcuxmx, 2021-07-16:
I think I've found the cause of the problem. Function mac_filter_delete_long() to remove the elements from the wrong linked list.

I changed ns_list_remove(&this->**short_filter_list**, filter_ptr); (line 244)to ns_list_remove(&this->**long_filter_list**, filter_ptr);，it works well.

https://github.com/ARMmbed/mbed-os/blob/d147abc3e556c58e5e343d34b729bc2192e18bd3/connectivity/nanostack/sal-stack-nanostack/source/MAC/IEEE802_15_4/mac_filter.c#L235-L248

0xc0170, 2021-07-16:
@artokin Please review