ARMmbed/mbed-os #14930: Is there a command to list detect codes of all supported mbed targets ?

mbedNoobNinja, 2021-07-19:
### Description of defect
Is there a command in the mbed's tool set, that can lists the detect codes for all mbed targets. I am adding a custom target and I want to make sure the custom detect code is unique and not used by some existing target. I know this info exists for targets supporting mbed v6.0 and above (it is located in targets.json), but targets before v6.0, are still recognised and their detect codes I don't know where to find.

I know that, with every new version of mbed, there might have new targets and their detect codes can match mine, but I was hoping when that happen, I can just run that magic comand and choose a new unique for my target and not searchng in the targets.json by hand.

#### Target(s) affected by this defect ?
N/A

#### Toolchain(s) (name and version) displaying this defect ?
N/A

#### What version of Mbed-os are you using (tag or sha) ?

master

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
**`mbed-cli 1.10.5`**

#### How is this defect reproduced ? 

N/A

Patater, 2021-07-20:
Hi @mbedNoobNinja

You can use [the Mbed CLI 2 board database snapshot json file](https://github.com/ARMmbed/mbed-tools/blob/master/src/mbed_tools/targets/_internal/data/board_database_snapshot.json) to see detect/product codes.

```
curl 'https://raw.githubusercontent.com/ARMmbed/mbed-tools/master/src/mbed_tools/targets/_internal/data/board_database_snapshot.json' \
| grep "product_code" \
| sort \
| uniq
```

jeromecoutant, 2021-07-20:
````
mbedls -l
````

mbedNoobNinja, 2021-07-21:
Nice, thanks guys, both methods will do the job.

Patater, 2021-07-22:
Glad to hear it's all good. I'll close this issue. Feel free to open another issue if you have other concerns.