ARMmbed/mbed-os #14933: Post binary hooks should support multiple executables

LDong-Arm, 2021-07-19:
### Description of defect

Currently, post binary hooks we define for CMake can be only run on one executable, which is assumed to be the top-level project (in `${CMAKE_BINARY_DIR}`). As we are adding all greentea tests as subdirectories of mbed-os, the hooks need to be able to run once per test.

#### Target(s) affected by this defect ?

Any target with a post binary hook (e.g. ARM_MUSCA_S1)

#### Toolchain(s) (name and version) displaying this defect ?

Any

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.13.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Any

#### How is this defect reproduced ? 

Compile a greentea test (e.g. using #14892), and the hook is not run on the binary. For example, a raw binary of the ticker test is ~40KB but a full binary for Musca S1 is ~1MB incl bootloader and TF-M secure binary.

LDong-Arm, 2021-07-20:
Because every CMake executable target has its own output binary path, a post build command needs to have the binary path as an argument. Having read as much CMake documentation and tried as many methods I can think of, I've come up with two
different implementations:
* Each Mbed targets (that has post-build operations) defines a _function_ which generates post build commands. The build system calls this function once per executable. (https://github.com/LDong-Arm/mbed-os/commit/4d66bac2b80af7a27f9b2a6ff0f3e239bc1028a4)
* Each Mbed target (that has post-build operations) defines a post build command _template_ with _placeholders_. The build system derives a post build command for each executable. (https://github.com/LDong-Arm/mbed-os/commit/2683dd03ee9a88be71f0dd693abda0d367452a2c)

Both works okay, but neither looks simple due to what CMake can do.

@ARMmbed/mbed-os-core What do you think?

Patater, 2021-07-21:
Could a CMake file from mbed-os (e.g. app.cmake) provide a function that applications can call and pass in their binary path to?

LDong-Arm, 2021-07-21:
After discussing with @Patater and @rwalton-arm, we decided to proceed with the first approach -  let Mbed target define a function that generates post-build commands, given a CMake executable target. PR raised to @rwalton-arm's fork for review: https://github.com/rwalton-arm/mbed-os/pull/1. Once merged (actually rebased and fast-forwarded), it'll automatically become part of #14892.

PR #14953 (2021-07-23): Rework post-build to support multiple executables

### Summary of changes 

The current implementation of post-build operations always assumes there's only one CMake executable target, at the root of the build directory. When a project creates multiple executables under different subdirectories, the target-specific post-build command fails to locate each built executable to act on.

This PR reworks the post-build mechanism to allow running the command once per executable target with correct path to executable. For details, refer to the commit message of "_Rework post-build to support multiple  [... cut]
closed 2021-07-23 by 0xc0170