ARMmbed/mbed-os #14934: Projects declared in wrong orders in mbed-os-examples-*

LDong-Arm, 2021-07-19:
### Description of defect

Most Mbed official examples (e.g. mbed-os-example-blinky) has the app's `project()` declared _after_ `add_subdirectory(mbed-os)`. This results in CMake treating `mbed-os` as the top-level project, rather than the app itself.

Despite the error, compilation happened to work fine, until we did some rework in #14892. We need to fix all the examples before being able to merge the aforementioned PR.

#### Target(s) affected by this defect ?

Any

#### Toolchain(s) (name and version) displaying this defect ?

Any

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.13.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Any

#### How is this defect reproduced ? 

Checkout #14892, and official examples can't compile because unit tests get enabled when the top-level is mbed-os, but it shouldn't be

LDong-Arm, 2021-07-20:
@rwalton-arm Thanks for fixing examples. We also need to fix the template in mbed-tools: https://github.com/ARMmbed/mbed-tools/blob/master/src/mbed_tools/project/_internal/templates/CMakeLists.tmpl

rwalton-arm, 2021-07-20:
> @rwalton-arm Thanks for fixing examples. We also need to fix the template in mbed-tools: https://github.com/ARMmbed/mbed-tools/blob/master/src/mbed_tools/project/_internal/templates/CMakeLists.tmpl

Was already on my list to fix, didn't add it here because this issue only talks about examples.

Patater, 2021-07-22:
All examples fixed, reviewed, and merged.

closed 2021-07-22 by Patater