ARMmbed/mbed-os #14936: `mbed compile` fails during post-build process

scottc11, 2021-07-20:
### Description of defect

`mbed compile` fails during post-build process

#### Target(s) affected by this defect ?

NUCLEO_F446RE

#### Toolchain(s) (name and version) displaying this defect ?

GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?
`6.13.0`

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed `1.10.5`

#### How is this defect reproduced ? 

run `mbed compile`

```
[DEBUG] Errors: /usr/local/Cellar/arm-none-eabi-gcc/9-2019-q4-major/gcc/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: BUILD/NUCLEO_F446RE/GCC_ARM/mbed-os/cmsis/device/rtos/TOOLCHAIN_GCC_ARM/mbed_boot_gcc_arm.o: in function `__wrap_main':
[DEBUG] Errors: /Users/me/Desktop/mbed-temp/./mbed-os/cmsis/device/rtos/TOOLCHAIN_GCC_ARM/mbed_boot_gcc_arm.c:77: undefined reference to `main'
[DEBUG] Errors: collect2: error: ld returned 1 exit status
Traceback (most recent call last):
  File "/Users/me/Desktop/mbed-temp/mbed-os/tools/make.py", line 74, in wrapped_build_project
    bin_file, update_file = build_project(
  File "/Users/me/Desktop/mbed-temp/mbed-os/tools/build_api.py", line 621, in build_project
    res = toolchain.link_program(resources, build_path, name)
  File "/Users/me/Desktop/mbed-temp/mbed-os/tools/toolchains/mbed_toolchain.py", line 780, in link_program
    self.link(elf, objects, libraries, lib_dirs, linker_script)
  File "/Users/me/Desktop/mbed-temp/mbed-os/tools/toolchains/gcc.py", line 356, in link
    self.default_cmd(cmd)
  File "/Users/me/Desktop/mbed-temp/mbed-os/tools/toolchains/mbed_toolchain.py", line 832, in default_cmd
    raise ToolException(stderr)
tools.utils.ToolException: /usr/local/Cellar/arm-none-eabi-gcc/9-2019-q4-major/gcc/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: BUILD/NUCLEO_F446RE/GCC_ARM/mbed-os/cmsis/device/rtos/TOOLCHAIN_GCC_ARM/mbed_boot_gcc_arm.o: in function `__wrap_main':
/Users/me/Desktop/mbed-temp/./mbed-os/cmsis/device/rtos/TOOLCHAIN_GCC_ARM/mbed_boot_gcc_arm.c:77: undefined reference to `main'
collect2: error: ld returned 1 exit status

[mbed] ERROR: "/usr/local/opt/python@3.9/bin/python3.9" returned error.
       Code: 1
       Path: "/Users/me/Desktop/mbed-temp"
       Command: "/usr/local/opt/python@3.9/bin/python3.9 -u /Users/me/Desktop/mbed-temp/mbed-os/tools/make.py -t GCC_ARM -m NUCLEO_F446RE --source . --build ./BUILD/NUCLEO_F446RE/GCC_ARM -v"
```

mbedmain, 2021-07-20:
@scottc11 thank you for raising this issue.Please take a look at the following comments:

Could you add some more detail to the description? A good description should be at least 25 words.

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

scottc11, 2021-07-20:
oh.... could this be because I don't have a main.cpp file yet?
[image]

scottc11, 2021-07-20:
It is obvious to me now that I was missing the `main.cpp` file in the root folder of the project.

I did an `mbed export -i vscode_gcc_arm` to use in VSCode and I guess somewhere during that process it left out `main.cpp`?

Or perhaps running `mbed new .` doesn't automatically generate a main.cpp file 🤷‍♂️

closed 2021-07-20 by scottc11