ARMmbed/mbed-os #14954: PermissionError: [Errno 13] Permission denied: 'folder' when using --custom-targets-json in mbed cli 2

mbedNoobNinja, 2021-07-22:
### Description of defect

PermissionError: [Errno 13] Permission denied: 'folder containing custom_targets.json' when trying to use custom_targets.json in different directory than project's root one.

#### Target(s) affected by this defect ?
N/A

#### Toolchain(s) (name and version) displaying this defect ?
N/A

#### What version of Mbed-os are you using (tag or sha) ?

master

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

**`mbed-tools 7.28.1`**

#### How is this defect reproduced ? 

`mbed-tools compile --custom-targets-json mbed-vk-boards -t GCC_ARM -m VK_RZ_A1H`

mbedmain, 2021-07-22:
@mbedNoobNinja thank you for raising this issue.Please take a look at the following comments:

Could you add some more detail to the description? A good description should be at least 25 words.

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

Patater, 2021-07-22:
Have you tried passing the path to the json file instead of the path to a folder? I suspect mbed-tools is trying to open the folder as a file and getting permission denied.

Something like this: `mbed-tools compile --custom-targets-json mbed-vk-boards/custom_targets.json -t GCC_ARM -m VK_RZ_A1H`

mbedNoobNinja, 2021-07-23:
Thank you @Patater, that was the problem now it compiles successfully. Close the Issue

closed 2021-07-23 by mbedNoobNinja