ARMmbed/mbed-os #14955: Does Cmake work with ARM compiler or just for some boards

mbedNoobNinja, 2021-07-23:
### Description of defect

Build fails with message: `armclang: error: ARM Compiler does not support '-march='` when compiling custom board with ARM compiler.

I noticed when compiling with GCC steps **`-- Check for working C compiler:`** are skipped. How to skip those steps for ARM compiler too.

```
mbed-tools compile --custom-targets-json mbed-vk-boards\custom_targets.json -t ARM -m VK_RZ_A1H
Configuring project and generating build system...
-- The C compiler identification is ARMClang 6.15.2
-- The CXX compiler identification is ARMClang 6.15.2
-- The ASM compiler identification is ARMClang
-- Found assembler: C:/Users/VEKATECH/AppData/Local/Mbed Studio/mbed-studio-tools/ac6/bin/armclang.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: C:/Users/VEKATECH/AppData/Local/Mbed Studio/mbed-studio-tools/ac6/bin/armclang.exe
-- Check for working C compiler: C:/Users/VEKATECH/AppData/Local/Mbed Studio/mbed-studio-tools/ac6/bin/armclang.exe - broken
CMake Error at C:/Python38/Lib/site-packages/cmake/data/share/cmake-3.20/Modules/CMakeTestCCompiler.cmake:66 (message):
  The C compiler

    "C:/Users/VEKATECH/AppData/Local/Mbed Studio/mbed-studio-tools/ac6/bin/armclang.exe"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: D:/V/MBED/cmake_build/VK_RZ_A1H/develop/ARM/CMakeFiles/CMakeTmp

    Run Build Command(s):C:/Python38/Scripts/ninja.exe cmTC_ab14d && [1/2] Building C object CMakeFiles\cmTC_ab14d.dir\testCCompiler.o
    FAILED: CMakeFiles/cmTC_ab14d.dir/testCCompiler.o
    C:\Users\VEKATECH\AppData\Local\MBEDST~1\MBED-S~1\ac6\bin\armclang.exe --target=arm-arm-none-eabi   -mthumb -Wno-armcc-pragma-push-pop -Wno-armcc-
pragma-anon-unions -Wno-reserved-user-defined-literal -Wno-deprecated-register -fdata-sections -fno-exceptions -fshort-enums -fshort-wchar --ide=mbed
-mfpu=vfpv3 -mfloat-abi=hard -mcpu=cortex-a9  -march= -MD -MT CMakeFiles\cmTC_ab14d.dir\testCCompiler.o -MF CMakeFiles\cmTC_ab14d.dir\testCCompiler.o.
d -o CMakeFiles\cmTC_ab14d.dir\testCCompiler.o -c testCCompiler.c
    armclang: error: ARM Compiler does not support '-march='
    ninja: build stopped: subcommand failed.

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  mbed-os/tools/cmake/app.cmake:24 (enable_language)
  CMakeLists.txt:13 (include)

-- Configuring incomplete, errors occurred!
See also "D:/V/MBED/cmake_build/VK_RZ_A1H/develop/ARM/CMakeFiles/CMakeOutput.log".
See also "D:/V/MBED/cmake_build/VK_RZ_A1H/develop/ARM/CMakeFiles/CMakeError.log".
ERROR: CMake invocation failed!

More information may be available by using the command line option '-v'.
```

#### Target(s) affected by this defect ?

`VK_RZ_A1H` (custom target) 

#### Toolchain(s) (name and version) displaying this defect ?

`ARM`

#### What version of Mbed-os are you using (tag or sha) ?

`mbed-os-6.12.0`

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
`python 3.8.7`
`cmake 3.20.5`
`mbed-tools 7.28.1`

#### How is this defect reproduced ? 

`mbed-tools compile --custom-targets-json mbed-vk-boards\custom_targets.json -t ARM -m VK_RZ_A1H`

0xc0170, 2021-07-23:
ARMClang is supported by CMake and mbed cli 2.

From the log above, `-march= ` is empty but -mcpu=cortex-a9 is set. How does this differ if you use normal rz_a1h board instead, same failure? From the recent CI run http://mbed-os-ci-public.s3-website-eu-west-1.amazonaws.com/?prefix=jenkins-ci/ARMmbed/mbed-os/mbed-os-ci/PR-14940/artifacts/14940/2/cmake-example-ARM/PASS/RZ_A1H/ - RZ_A1H passed. Is this an issue with a custom target - march flag is being not set properly?

Some details for mcpu/march:

> On Arm, this flag is a combination of -march and -mtune
Taken from https://community.arm.com/developer/tools-software/tools/b/tools-software-ides-blog/posts/compiler-flags-across-architectures-march-mtune-and-mcpu

This is the logic for ARMClang CMake https://github.com/Kitware/CMake/blob/master/Modules/Compiler/ARMClang.cmake#L98 (how it's appending march/mcpu).  We define `CMAKE_SYSTEM_PROCESSOR` only in our CMake (see mbed_toolchain.cmake).

LDong-Arm, 2021-07-23:
In Mbed OS, we currently set `CMAKE_SYSTEM_PROCESSOR` which CMake translates into `-mcpu=${CMAKE_SYSTEM_PROCESSOR}`. If I remember correctly, CMake adds `-march` if `CMAKE_SYSTEM_ARCH` is set - is there any chance you have this variable in your local project?

It was a known issue of CMake that `CMAKE_SYSTEM_ARCH` and `CMAKE_SYSTEM_PROCESSOR` are pretty unpredictable and hard to deal with (before version 3.21.0). Starting from CMake 3.21.0, it's possible/recommended to not use those two variables anymore. PR #14575 (close to getting merged) is to take advantage of this, which should remove the issue if you have the very latest CMake.

After getting pass this issue, a full version of Arm Compiler should be able to compile Mbed OS. But the free version from Mbed Studio, which you are using, has a restriction which will fail CMake's compiler check: CMake builds _its own_ dummy application to test if the compiler and linker work, but the free version of Arm linker `armlink` doesn't allow the final executable to not contain any "mbed" symbols (but CMake's built-in dummy test app is not an Mbed app). Even if `--ide=mbed` is passed to `armlink`, this symbol restriction still applies. At this point CMake doesn't offer a way to skip this check, but hopefully we'll find a solution/workaround at some point.

mbedNoobNinja, 2021-07-23:
@0xc0170 I can confirm, the same behavior is observed with RZ_A1H board. None of the CmakeLists.txt files located in the root folder & custom-target's folder has a line setting `-march` flag, so as @LDong-Arm mentioned it must come from the CMake itself.

I am currently using cmake installed as python module and found where ARMClang.cmake is located, but have no idea how to bypass this `-march` flag insertion ... CMake is not my thing.

LDong-Arm, 2021-07-23:
> I am currently using cmake installed as python module and found where ARMClang.cmake is located, but have no idea how to bypass this `-march` flag insertion ... CMake is not my thing.

You can upgrade CMake to 3.21.0 (https://cmake.org/download/) _and_ use PR #14575 (or rebase any changes onto that). Those two combined should resolve the `-march` issue.

mbedNoobNinja, 2021-07-23:
... and again @LDong-Arm was correct, linking fails even **`-march`** is gone ...

```
CMake Error at C:/Python38/Lib/site-packages/cmake/data/share/cmake-3.20/Modules/CMakeTestCCompiler.cmake:66 (message):
  The C compiler

    "C:/Users/VEKATECH/AppData/Local/Mbed Studio/mbed-studio-tools/ac6/bin/armclang.exe"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: D:/MBED/cmake_build/RZ_A1H/develop/ARM/CMakeFiles/CMakeTmp

    Run Build Command(s):C:/Python38/Scripts/ninja.exe cmTC_e5cb6 && [1/2] Building C object CMakeFiles\cmTC_e5cb6.dir\testCCompiler.o
    [2/2] Linking C executable cmTC_e5cb6.elf
    FAILED: cmTC_e5cb6.elf
    cmd.exe /C "cd . && C:\Users\VEKATECH\AppData\Local\MBEDST~1\MBED-S~1\ac6\bin\armlink.exe    CMakeFiles\cmTC_e5cb6.dir\testCCompiler.o -o cmTC_e5c
b6.elf  --ide=mbed --list=cmTC_e5cb6.elf.map && cd ."
    Fatal error: L6048U: Invalid input. This variant of Arm Compiler is licensed only for use with Mbed Studio.
    Finished: 0 information, 0 warning, 0 error and 1 fatal error messages.
    ninja: build stopped: subcommand failed. 
```

... well I guess I should live with it until the workaround is merged. Thank you both. This can be closed.

LDong-Arm, 2021-07-23:
> This variant of Arm Compiler is licensed only for use with Mbed Studio.

@mbedNoobNinja Now you've hit the license restriction.

I've raised #14957, if you try this PR (and also use CMake 3.**21**.0), it should compile.

LDong-Arm, 2021-07-23:
@mbedNoobNinja 
Update: Windows supported added to #14957, feel free to have a try.

LDong-Arm, 2021-08-02:
Actually Mbed Studio's Arm Compiler license seems not available as a standalone file on Windows, in which case the compiler can only be triggered from Mbed Studio.

> ... well I guess I should live with it until the workaround is merged. Thank you both. This can be closed.

@mbedNoobNinja Thanks. I'm going to close it.

closed 2021-08-02 by LDong-Arm