ARMmbed/mbed-os #14958: Silicon Labs EFM32 - linker fail on Thunderboard Sense 2 when compiling from PlatformOI.

giedrius-stanevicius, 2021-07-23:
Linker fails with error: "cannot find -lrail_efr32xg12_release.ar"

Original issue is reported at https://github.com/platformio/platform-siliconlabsefm32/issues/4

I suspect that files `rail_efr32xg12_release.ar` and `rail_efr32xg12_release.a` should be placed in separate directories according supported toolchains.

#### Target(s) affected by this defect ?
TB_SENSE_12

#### Toolchain(s) (name and version) displaying this defect ?
GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.9.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ? 

See https://github.com/platformio/platform-siliconlabsefm32/issues/4

mbedmain, 2021-07-23:
@giedrius-stanevicius thank you for raising this issue.Please take a look at the following comments:

Could you add some more detail to the description? A good description should be at least 25 words.
It would help if you could also specify the versions of any tools you are using?
 
NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

0xc0170, 2021-07-27:
@ARMmbed/team-silabs Please review