ARMmbed/mbed-os #14968: Upgrade our testing systems to use GNU Arm Embedded Toolchain 10-2021-q2-update as the new supported version of `gcc`

Patater, 2021-07-29:


Patater, 2021-08-19:
I've raised https://github.com/ARMmbed/mbed-os/pull/15007 to move our GitHub actions CI to the new version of GCC.

@saheerb has been working on updating the Jenkins CI.

LDong-Arm, 2021-08-25:
It's slightly annoying that Mbed CLI 1 prints
```
[Warning] @,: Compiler version mismatch: Have 10.3.1; expected version >= 9.0.0 and < 10.0.0
```
The version range is hardcoded in tools/toolchains/gcc.py which is frozen. As it's just a warning, leaving it as it is probably fine...

jeromecoutant, 2021-09-06:
> As it's just a warning, leaving it as it is probably fine...

I would prefer to remove warnings
Thx

rwalton-arm, 2021-09-06:
> It's slightly annoying that Mbed CLI 1 prints
> 
> ```
> [Warning] @,: Compiler version mismatch: Have 10.3.1; expected version >= 9.0.0 and < 10.0.0
> ```
> 
> The version range is hardcoded in tools/toolchains/gcc.py which is frozen. As it's just a warning, leaving it as it is probably fine...

I don't think fixing that warning would affect the behaviour of the online build system. We should be OK to fix it without creating a new "tools release" for the online compiler.

LDong-Arm, 2021-09-06:
Ultimately we encourage users to move away from Mbed CLI 1. But if someone disregards the deprecation and does what the warning says (i.e. downgrade GCC to 9.x), they may encounter issues such as NU_M2354 unable to run due to known issues of the old GCC with Armv8-m.

Patater, 2021-09-06:
> We should be OK to fix it without creating a new "tools release" for the online compiler.

Agreed. The warning is annoying and not helpful for users, and ignored by the online compiler which doesn't use gcc anyway.

0xc0170, 2021-09-06:
I'll send the PR now to remove the warning for ver 10 as only two PRs left to be merged (docs and docker image update).

0xc0170, 2021-09-06:
Already removed in the docker image update https://github.com/ARMmbed/mbed-os/pull/15007/commits/44e579524d5e7aa210784641d5457238b8a9a97e

Patater, 2021-09-07:
Saheer has updated Jenkins CI.

We've also merged https://github.com/ARMmbed/mbed-os/pull/15007 which adds support in GitHub actions via our Docker images.

closed 2021-09-07 by Patater

jeromecoutant, 2021-09-07:
> Already removed in the docker image update [44e5795](https://github.com/ARMmbed/mbed-os/commit/44e579524d5e7aa210784641d5457238b8a9a97e)

Patch is strange... only GCC_RANGE has to be updated ?

LDong-Arm, 2021-09-07:
> Patch is strange... only GCC_RANGE has to be updated ?

[ARMCC_RANGE](https://github.com/ARMmbed/mbed-os/blob/master/tools/toolchains/arm.py#L549_ is 6.10 to 7.0 so we won't need to worry about it in a long while.