ARMmbed/mbed-os #14971: ThisThread::sleep_for never returns on Nucleo-F401RE with external crystal and bare-metal

riaancillie, 2021-07-31:
### Description of defect

When running a bare-metal profile on a STM32F401RE and using an external crystal, ThisThread::sleep_for never returns. This was tested on both a custom PCB and the Nucleo F401RE with a 8Mhz crystal soldered on. The external crystal definitely works since setting BOOT0=1 creates a working DFU USB device.

#### Target(s) affected by this defect ?
STM32F401RE / Nucleo-F401RE

#### Toolchain(s) (name and version) displaying this defect ?
Mbed-studio with GCC Compiler

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os-6.10.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Mbed Studio v1.4.1

#### How is this defect reproduced ? 

main.cpp:
```
#include <mbed.h>

DigitalOut led(LED1);

int main()
{
  led = !led;
  ThisThread_sleep_for(500ms);  //This causes everything to freeze
  /*
  //Using old-school delay method works:
  for (int i = 0; i < 500; i++)
    wait_us(1000);
  */
}
```

mbed_app.json
```
{
    "requires": ["bare-metal","drivers-usb","events"],
    "target_overrides": {
        "*": {
          "target.c_lib": "std",
          "target.printf_lib": "minimal-printf",
          "platform.minimal-printf-enable-floating-point": true,
           "platform.minimal-printf-set-floating-point-max-decimals": 6          
        },
        "NUCLEO_F401RE": {                    
          "target.device_has_add": ["USBDEVICE"],
          "target.clock_source": "USE_PLL_HSE_XTAL"
        }
      }
}
```

0xc0170, 2021-08-02:
cc @ARMmbed/team-st-mcd

jeromecoutant, 2021-08-16:
Hi
For information, does it work without USB support addition?
Thx

riaancillie, 2021-08-16:
> Hi
> For information, does it work without USB support addition?
> Thx

No, simply using a bare-metal profile alone with no extra components added still results in the same behavior.

jeromecoutant, 2021-10-01:
Maybe a `while(true)` loop is missing in main() ?

0xc0170, 2022-07-22:
As it's been almost a year without any update, I'll close this as won't fix.

closed 2022-07-22 by 0xc0170