ARMmbed/mbed-os #14974: BLE: Features/roles should be disabled by default

boraozgen, 2021-08-02:
### Description of defect

I noticed that my BLE application, which only uses the broadcaster/peripheral roles and GATT server feature was taking up ~300kB of flash in debug build. When I looked into it a bit more, I noticed that many functions which I don't use are linked into the binary. I could reduce the binary size by ~80kB by disabling the features in mbed_app.json. I believe it is a bad default behavior to link all of the unused code. Therefore I suggest disabling all of the features/roles by default and only enabling the required ones in the examples. This way users would build the smallest binary necessary.

#### Target(s) affected by this defect ?
BLE targets.

#### Toolchain(s) (name and version) displaying this defect ?
Not related.

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-99.99.99

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Not related.

#### How is this defect reproduced ? 
Design issue.

jeromecoutant, 2021-08-16:
@pan-

0xc0170, 2021-08-16:
@ARMmbed/mbed-os-connectivity Please review

boraozgen, 2021-11-29:
Disabling features may have side effects: https://github.com/ARMmbed/mbed-os/issues/15182

ladislas, 2021-12-01:
> @boraozgen Would it be helpful to make the documentation more explicit about disabling unnecessary features?

Answering for our project: Yes, that would be helpful. We are not running out of memory yet, but optimizing code size is always a good thing. Plus the best bug free code is the code that just ain't there.

> I think if we do make this change we must cover each configuration parameter in detail in the documentation to make sure people know what they get when enabling/disabling each. Further, some of the features rely on other ones being enabled (eg: periodic advertising/extended advertising) so those dependencies should also be covered in updated documentation.

I think that's a must do if we want mbed's BLE features to be used in real life devices. At some point people are going to ask questions, just like we do here, and having the documentation sooner than later will make sure people find what they need and do switch to Zephyr or something else because mbed's BLE seems to opaque because of poor documentation.