ARMmbed/mbed-os #14977: Upgrade to TF-M v1.4

Patater, 2021-08-04:


LDong-Arm, 2021-08-31:
Dependencies:
* #15029
* #15027
* https://github.com/ARMmbed/mbed-os-tf-m-regression-tests/pull/125

Notes
* Update TF-M version in targets/TARGET_NUVOTON/scripts/NUVOTON.py

ccli8, 2021-09-01:
> As a convention, the prebuilt images committed into Mbed OS is built by the Arm Compiler.

Would you build for M2354 with GCC (10.3)? Our company doesn't license latest Arm (6.16) compiler.

> Update hardcoded TF-M version in targets/TARGET_NUVOTON/scripts/NUVOTON.py

And also in `tools/targets/NU_M2354.py`.

LDong-Arm, 2021-09-01:
@ccli8 I'll edit the list to use GCC 10.3 for NU_M2354.

On our side, we'll build with both GCC 10.3 and Arm Compiler 6.16 to make sure compilation works.

LDong-Arm, 2021-09-01:
@ccli8 We've found an NU_M2354 board so we'll be able to test it ourselves first. But we'll let you know our pull requests, thanks in advance.

closed 2021-09-15 by Patater