ARMmbed/mbed-os #14987: CMake project definitions for cloud examples are incorrect

rwalton-arm, 2021-08-09:
### Description of defect
Same as #14934 but specific to "cloud examples".

### Examples to fix
- [ ] https://github.com/ARMmbed/mbed-os-example-for-aws/pull/49
- [ ] https://github.com/ARMmbed/mbed-os-example-for-google-iot-cloud/pull/32
- [ ] https://github.com/ARMmbed/mbed-os-example-for-azure/pull/34

mbedmain, 2021-08-09:
@rwalton-arm thank you for raising this issue.Please take a look at the following comments:

Could you add some more detail to the description? A good description should be at least 25 words.
What target(s) are you using?
What toolchain(s) are you using?
What Mbed OS version are you using?
It would help if you could also specify the versions of any tools you are using?
 How can we reproduce your issue?

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

Patater, 2021-08-09:
Issue fixed via linked merged PRs.

closed 2021-08-09 by Patater