ARMmbed/mbed-os #14992: Remove "tools update" from .github/workflows/greentea_cmake.yml

rwalton-arm, 2021-08-11:
### Description of defect
To be done when `mbed-os-env` updates to mbed-tools version 7.30.1 or later

#### Target(s) affected by this defect ?

#### Toolchain(s) (name and version) displaying this defect ?

#### What version of Mbed-os are you using (tag or sha) ?

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ?

mbedmain, 2021-08-11:
@rwalton-arm thank you for raising this issue.Please take a look at the following comments:

Could you add some more detail to the description? A good description should be at least 25 words.
What target(s) are you using?
What toolchain(s) are you using?
What Mbed OS version are you using?
It would help if you could also specify the versions of any tools you are using?
 How can we reproduce your issue?

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

rwalton-arm, 2021-08-11:
We were using the incorrect docker image. Closing.

closed 2021-08-11 by rwalton-arm