ARMmbed/mbed-os #14996: Question: How to get the best parameters of read command in QSPIFBlockDevice Class？

Berte-Yu, 2021-08-12:
### Description of defect

  I want to get the best read parameters in SFDP through `QSPIFBlockDevice`, such as read_instruction, read_dummy_cycles, read_alt_size,etc, But some related methods are set to static

#### Target(s) affected by this defect ?

#### Toolchain(s) (name and version) displaying this defect ?

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.13.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ?

mbedmain, 2021-08-12:
@Berte-Yu thank you for raising this issue.Please take a look at the following comments:

What target(s) are you using?
What toolchain(s) are you using?
It would help if you could also specify the versions of any tools you are using?
 How can we reproduce your issue?

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

0xc0170, 2021-08-12:
I don't understand what are: the best read parameters ? `QSPIFBlockDevice` provides block device functionlity like read/write or erase blocks. The members like instruction, cycles etc are private. why would you want them to be exposed? You can turn on tracing to have these printed when being executed.

Berte-Yu, 2021-08-13:
QSPI flash support:
* Read Data  1-1-1
* Fast Read Dual I/O 1-2-2
* Fast Read Quad I/O 1-4-4
the bast read Instruction is `Fast Read Quad I/O`, this Instruction required dummy cycle, mode cycle, address width data_width,etc. called the best read parameters.

In my MCU, it support CACHE module, it can automatic read my code data from QSPI_FLASH. So I need to configure these parameters in this module to tell CACHE how to read data from QSPI_FLASH。

Class `QSPIFBlockDevice` support SFDP, so  I wanted to use it directly without having to implement SFDP methods again

0xc0170, 2021-08-25:
If you need some customization, you need to create your own blockdevice.

But in this case, there are methods like `_sfdp_detect_best_bus_read_mode` that should check and set the best read mode. Form the comment in the code `    // Detect and Set fastest Bus mode (default 1-1-1)`