ARMmbed/mbed-os #14999: Hard fault on DISCO F769NI that occurs when STM Ethernet driver is initialized

mikaleppanen, 2021-08-13:
### Description of defect

We are seeing hard faults on DISCO F769NI that occur when STM Ethernet driver is initialized on https://github.com/PelionIoT/pelion-border-router 

The error occurs sometimes during startup and is hard to repeat. When error happens the console gets stuck and no hard fault trace is printed by Mbed OS.

Error can be repeated by running board startup of reset loop using mbed system_reset() where the board is set to reset after the Ethernet driver is initialized. Sometimes the initialization fails to hard fault before the reset.

When running under debugger the fault happens when SCB_DisableDCache() loop for "clean & invalidate D-Cache" is ongoing and UART interrupt happens.

Here is a screen shot:

[image]

Hard fault cannot be repeated if interrupts are disabled before calling disable data cache call here:

https://github.com/ARMmbed/mbed-os/blob/4dd08c40f557f3930c0c717b39c6cb48aba177b9/connectivity/drivers/emac/TARGET_STM/TARGET_STM32F7/TARGET_DISCO_F769NI/stm32f7_eth_init.c#L41 

by 

```
core_util_critical_section_enter();
SCB_DisableDCache();
core_util_critical_section_exit();
```

#### Target(s) affected by this defect ?

DISCO F769NI

#### Toolchain(s) (name and version) displaying this defect ?

arm-none-eabi-gcc.exe (GNU Arm Embedded Toolchain 9-2020-q2-update) 9.3.1 20200408 (release)

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.12.0
mbed-os-6.12.0-rc1

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

None

#### How is this defect reproduced ? 

To compile pelion-border-router:
```
git clone git@github.com:PelionIoT/pelion-border-router.git
cd pelion-border-router
mbed deploy
mbed compile -t GCC_ARM -m DISCO_F769NI
```
binary is pelion-border-router.bin

mbed system_reset() can be called e.g. after trace 

https://github.com/ARMmbed/mbed-os/blob/e588f80d13f288804af1682d701887cd831abcc5/connectivity/drivers/emac/TARGET_STM/stm32xx_emac.cpp#L363

cloud credentials are not needed, this #error can be commented out on 

https://github.com/PelionIoT/pelion-border-router/blob/7a4f41f65fbedf8e09271989c1e16d7eded700fc/mbed_cloud_dev_credentials.c#L21

Radio shields etc. are not needed in addition to DISCO_F769NI to test the Ethernet initialization.

0xc0170, 2021-08-13:
cc @ARMmbed/team-st-mcd

LMESTM, 2021-08-13:
@mikaleppanen  your proposed fix seems to be valid at first glance, would you please send it as a pull request and this can be reviewed / approved by @jeromecoutant  ?

mikaleppanen, 2021-08-16:
> @mikaleppanen your proposed fix seems to be valid at first glance, would you please send it as a pull request and this can be reviewed / approved by @jeromecoutant ?

Ok, I'll make a pull request.

closed 2021-08-17 by 0xc0170