ARMmbed/mbed-os #15004: lorawan-1-1 Branch, use-secure-element not implemented?

niondir, 2021-08-16:
### Description of defect

I'm working with the lorawan-1-1 branch. There is a config parameter `use-secure-element` that seems not to be used.

Is there any plans or branch where secure element usage is setup for LoRaWAN?

#### Target(s) affected by this defect ?
all

#### Toolchain(s) (name and version) displaying this defect ?
n/a

#### What version of Mbed-os are you using (tag or sha) ?
 lorawan-1-1 branch

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
n/a

#### How is this defect reproduced ? 
n/a

0xc0170, 2021-08-16:
@ARMmbed/mbed-os-connectivity Please review

hallard, 2021-08-25:
I'm not sure this is relevant, I mean secure element can be used various different way and it's depending on application not on the stack. So from my point of view it's end user LoRaWAN application to manage this, but correct me if I'm wrong.

I mean for example:

- what secure element hardware is used?
- does it auth on join server? 
- does it has join server certificate on it and on which slot?
- does it need to be configured/programmed before?
- what we do with serial number of secure element?

As you can se we can use it for lot of things, so not sure implementing that is the stack is relevant, but I may forgot something.

niondir, 2021-09-04:
I think at least a documented way to exchange the underlying encryption and decryption is needed to use the secure element. 

For the devEUI and parameters I can already pass to the Stack I agree that no explicit support is needed and it's up to the application.

Is there already some API? Or would I need to change / override some code of the lorawan stack?