ARMmbed/mbed-os #15009: Deprecate device management in mbed-cli1

0xc0170, 2021-08-17:
### Description of defect

Reference: https://github.com/ARMmbed/mbed-os/pull/14979 - pelion team removed old requirements that are not relevant anymore. We missed a device management functionality in mbed-cli1: https://github.com/ARMmbed/mbed-cli/blob/master/mbed/mbed.py#L3036-L3046

Shall we deprecate this device command completely in the september release as it's not supported anymore. This is the up to date device management https://developer.pelion.com/docs/device-management/current/connecting/tutorial-pelion-mbedos.html

@teetak01 Is that correct? we should deprecate device management we provided via mbed-cli1 as it's not up to date, and has not been removed from Mbed OS.

#### Target(s) affected by this defect ?

Any

#### Toolchain(s) (name and version) displaying this defect ?

Any

#### What version of Mbed-os are you using (tag or sha) ?

Mbed OS 6.13

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Any

#### How is this defect reproduced ? 

https://github.com/ARMmbed/mbed-cli/blob/master/mbed/mbed.py#L3036-L3046 - device management part in our tools

teetak01, 2021-08-17:
@0xc0170 as far as I know, there is no plans to maintain DM-mode like functionality in CLI2, so we should just remove the remaining legacy features also from CLI1. Otherwise it would require some effort and changes to CLI1 to support manifest-tool v2.

All pelion tutorials have already moved away from using DM-mode (instead they use manifest-tool directly). The manifest-tool v2 provides the easy-to-use features for developers https://github.com/PelionIoT/manifest-tool#manifest-dev-tool

0xc0170, 2021-08-30:
I created https://github.com/ARMmbed/mbed-os-5-docs/pull/1480 to resolve this. As old tools are frozen, I updated the documentation regarding mbed-cli dm command - deprecate its functionality.