ARMmbed/mbed-os #15014: STM32G0B0XX not in ARM Pack manager

SibertEnovates, 2021-08-20:
### Description of defect
I'm trying to create a custom target using the new upcoming STM32G0B0-series chip, but I when I use `mbed export`, I get following errors:
`Configuration error: Bootloader not supported on this target. ROM start not found in targets.json.
Configuration error: Bootloader not supported on this target. RAM start not found in targets.json.`

I suspect this is a similar issue to https://github.com/ARMmbed/mbed-os/pull/13237

#### Target(s) affected by this defect ?
STM32G0B0

#### Toolchain(s) (name and version) displaying this defect ?
ARM-GCC

#### What version of Mbed-os are you using (tag or sha) ?
Tag: mbed-os-6.10.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
MBED CLI 1.10.4

#### How is this defect reproduced ? 
Create a new target for STM32G0B0 and a board using this chip. Execute `mbed export` to get the errors.

jeromecoutant, 2021-08-20:
Hi

2 solutions:
- make a patch like #14477
- update json file like #14975

PS: about you custom board, I advice you to make it public in https://github.com/ARMmbed/stm32customtargets

Regards,

SibertEnovates, 2021-08-20:
Hi @jeromecoutant 

I would like to know where I can find the right values or how I should determine the right values to be added to the ARM Pack manager for this chip. I can't seem to find anything related to sectors in the datasheet but then again I'm not very familiar with the matter

Thanks in advance,

jeromecoutant, 2021-08-20:
Information originally comes form CMSIS pack: https://www.keil.com/dd2/pack/
pack files are a zip file, check info in pdsc files

SibertEnovates, 2021-09-14:
We managed to get it working be replacing
```
LoopFillZerobss:
  cmp r2, r4
  bcc FillZerobss

/* Call static constructors */
  bl __libc_init_array
/* Call the application s entry point.*/
  bl main
```
by
```
LoopFillZerobss:
  cmp r2, r4
  bcc FillZerobss

/* Call the clock system intitialization function.*/
  bl  SystemInit
	bl _start
	bx lr
```

In the startup-file. The other G0 startup files in the Mbed repo seem to do it that way as well and thus don't match the startup files found in the STM32Cube CMSIS folder.

Can someone verify if this is correct?

jeromecoutant, 2021-09-14:
Yes!
This is what I am doing when I am updating mbed files from STM32Cube !

SibertEnovates, 2021-09-14:
@jeromecoutant thanks for the confimation. Would you mind providing an explanation why this change is necessary?

jeromecoutant, 2021-09-14:
Should be something like
https://os.mbed.com/docs/mbed-os/v6.14/program-setup/bootstrap.html

@ARMmbed/mbed-os-core

SibertEnovates, 2021-09-15:
@jeromecoutant I'm setting up a PR for the new target, but the template mentions the requirement 'For new targets please attach full test results for all supported compilers". I can't seem to find how to run these test. Can you point me in the right direction?

jeromecoutant, 2021-09-15:
Idea from ARM is , I think, to execute test commands like:
`mbed test -m <YOUR_TARGET> -t GCC_ARM`
@MarceloSalazar

You need a Mbed target Id in https://github.com/ARMmbed/mbed-os-tools/blob/master/src/mbed_os_tools/detect/platform_database.py

If you don't have, you can share your custom board in https://github.com/ARMmbed/stm32customtargets

About PR, I am also going to push very soon a STM32G0 update :-)
See https://github.com/jeromecoutant/mbed/tree/PR_G0_CUBE

SibertEnovates, 2021-09-15:
@jeromecoutant it seems a bit pointless to push my changes if you're working on an update for the STM32G0, which includes this chip. Unless you think it's usefull, I won't push my changes then?

jeromecoutant, 2021-09-15:
Are your changes working on top of my branch.....?

SibertEnovates, 2021-09-15:
https://github.com/SibertEnovates/mbed-os/tree/STM32G0B0xx

SibertEnovates, 2021-09-15:
My changes don't seem to overlap so I'll push them

jeromecoutant, 2021-09-15:
> https://github.com/SibertEnovates/mbed-os/tree/STM32G0B0xx

I think NUCLEO_G0B0RE doesn't exist...
I understand it the base for your custom board ?

Check #15075
You should be able to make your target like with
`        "inherits": [
            "MCU_STM32G0B0xE"
        ]`

jeromecoutant, 2021-10-05:
Hi
I think we can close this issue as STM32G0B0RETx is now part of  `tools/arm_pack_manager/index.json` ?
Thx

closed 2021-10-05 by 0xc0170