ARMmbed/mbed-os #15021: Unused function warning validate_errno_values

jimparis, 2021-08-24:
### Description of defect
Build warning:
```[32/220] Building CXX object CMakeFiles/main.dir/mbed-os/platform/source/mbed_retarget.cpp.obj
/home/jim/git/coris/fw/lora-bps/mbed-os/platform/source/mbed_retarget.cpp:1942:14: warning: 'void validate_errno_values(int)' defined but not used [-Wunused-function]
 static  void validate_errno_values(int value)
              ^~~~~~~~~~~~~~~~~~~~~
```

#### Target(s) affected by this defect ?
XDOT_L151CC

#### Toolchain(s) (name and version) displaying this defect ?
GCC_ARM
arm-none-eabi-gcc (15:8-2019-q3-1) 8.3.1 20190703 (release) [gcc-8-branch revision 273027]

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.14.0 3377f083b3a6

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
$ mbed-tools --version
7.32.0

#### How is this defect reproduced ? 
Building

mbedmain, 2021-08-24:
@jimparis thank you for raising this issue.Please take a look at the following comments:

Could you add some more detail to the description? A good description should be at least 25 words.

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

0xc0170, 2021-08-25:
Shall we add `MBED_UNUSED` to this function to fix the warning? @jimparis Can you send a pull request?