ARMmbed/mbed-os #15035: STM32WB55 Mac address changing every power cycle

denizagrali, 2021-08-31:
### Description of defect

#### Target(s) affected by this defect ?
STM32WB55

#### Toolchain(s) (name and version) displaying this defect ?
Mbed Studio: 1.4.1

#### What version of Mbed-os are you using (tag or sha) ?

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Latest

mbedmain, 2021-08-31:
@denizagrali thank you for raising this issue.Please take a look at the following comments:

Could you add some more detail to the description? A good description should be at least 25 words.
What Mbed OS version are you using?
How can we reproduce your issue?

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

0xc0170, 2021-09-01:
Closed as duplicate

closed 2021-09-01 by 0xc0170

LMESTM, 2021-09-01:
You may have a look at :
https://www.st.com/content/ccc/resource/technical/document/application_note/group1/43/ea/2f/dc/10/a3/46/e6/DM00598033/files/DM00598033.pdf/jcr:content/translations/en.DM00598033.pdf
and see if that can help.