ARMmbed/mbed-os #15036: STM32WB55 Mac address changing every power cycle

denizagrali, 2021-08-31:
### Description of defect

Hello , my issue is kind of  with the i think GAP API. I don't want the mac address to change but address is changing every power cycle. How can i make to be static this address?

#### Target(s) affected by this defect ?
STM32WB55

#### Toolchain(s) (name and version) displaying this defect ?
Mbed Studio: 1.4.1

#### What version of Mbed-os are you using (tag or sha) ?

    Mbed os 6.14.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Latest

mbedmain, 2021-08-31:
@denizagrali thank you for raising this issue.Please take a look at the following comments:

How can we reproduce your issue?

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

denizagrali, 2021-08-31:
> @denizagrali thank you for raising this issue.Please take a look at the following comments:
>
> How can we reproduce your issue?
>
> NOTE: If there are fields which are not applicable then please just add [...]

Thanks for your commit. I arranged my issue.

LMESTM, 2021-09-01:
Same or similar issues have already been discussed:
https://github.com/ARMmbed/mbed-os/issues/12115
https://github.com/ARMmbed/mbed-os/issues/13628

denizagrali, 2021-09-01:
Already i read those but because it has been updated there are changes and i dont understand clearly

denizagrali, 2021-09-01:
Ok, now I was able to create new mac address without any problem. Well how can i advertise this mac address ??

LMESTM, 2021-09-01:
You may have a look at :
https://www.st.com/content/ccc/resource/technical/document/application_note/group1/43/ea/2f/dc/10/a3/46/e6/DM00598033/files/DM00598033.pdf/jcr:content/translations/en.DM00598033.pdf
and see if that can help.

denizagrali, 2021-09-01:
Thanks for everything ,I solved it with your help :)

closed 2021-09-01 by 0xc0170