ARMmbed/mbed-os #15038: Can't get program working on NUCLEO_WL55JC

hallard, 2021-08-31:
### Description of defect

Nucleo WL55JC board is recognized and can be flashed without any issue but once flashed, any program does not seems to start, I tested blinky also without any success.

#### Target(s) affected by this defect ?

NUCLEO_WL55JC

#### Toolchain(s) (name and version) displaying this defect ?

GNU Arm Embedded Toolchain 10-2020-q4-major 10.2.1 20201103 (release)

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-99.99.99
    
#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed studio V1.4.1 
mbed-tools 7.27.0

#### How is this defect reproduced ? 

From mbed ide `New Program` then `mbed-os-example-blinky` then plug `NUCLEO_WL55JC` with USB and select as target (it's auto detected) then click on `run program` This compile, then flash, then nothing.

From mbed CLI2
```
mbed-tools import mbed-os-example-blinky
cd mbed-os-example-blinky
mbed-tools compile -m NUCLEO_WL55JC -t GCC_ARM --flash
```
This compile, then flash, then nothing either

Don't know if it's related #14185 or #14236 but if someone has this board in hand, would be interested by a try, I first thought it was related from my office NUCLEO WL55 board, but tested at home with another one (and another computer), same thing (both received last week from mouser).

I tried on old B-L072Z-LRWAN1 and worked like a charm, so I suspect something strange on WL55JC. 
Pay attention that I'm working on LoRa-E5 and RAK3172 (STM32WL Family also) without any issue but they are single core cortex M4, not dual core as WL55JC

jeromecoutant, 2021-09-01:
Same issue with another random board... (NUCLEO_L552ZE_Q) :-(

mbed-tools                    7.32.0

jeromecoutant, 2021-09-01:
This issue should maybe be duplicated to https://github.com/ARMmbed/mbed-tools/issues ?

@Patater @0xc0170

0xc0170, 2021-09-01:
It's fine to keep it here, I'll add this to the core project.

0xc0170, 2021-09-01:
> Same issue with another random board... (NUCLEO_L552ZE_Q) :-(

Can you git bisect to find when it broke?

jeromecoutant, 2021-09-01:
I can't find any working version :-(

It's working with DISCO_L475VG_IOT01A

hallard, 2021-09-01:
I wanted to try with mbed-cli (CLI1) so I plugged the nucelo on my USB cable and doing just nothing since yesterday, now the board is blinking, just amazing I did not flashed the board since last flash that was not working.

hallard, 2021-09-01:
@jeromecoutant did you updated the embedded STlink firmware of your Nucleo boards?

jeromecoutant, 2021-09-01:
> @jeromecoutant did you updated the embedded STlink firmware of your Nucleo boards?

Yes:
```
$ mbedtools detect
Board name     Serial number             Serial port    Mount point(s)    Build target(s)    Interface Version
-------------  ------------------------  -------------  ----------------  -----------------  -------------------
NUCLEO-WL55JC  003600235553500a20393256  COM8           D:                NUCLEO_WL55JC      V3J8M3
```

After more tests, I agree there is some random issues... I flash application several times and/or unplug/plug the boards. Sometimes blink is started, sometimes not... :-(

@SCHSTM

hallard, 2021-09-01:
> After more tests, I agree there is some random issues... I flash application several times and/or unplug/plug the boards. Sometimes blink is started, sometimes not... :-(

Yeah same thing here, to be sure it's not working, flash with mbed ide and board won't start, then flashed back with CLI2 works all time (today because it was not yesterday)

I'll try something later, build and flash with mbed ide then power the board with 3.3V only removing all USB/STlink Stage just to see.

jeromecoutant, 2021-10-01:
Got the same issue I think : https://github.com/ARMmbed/mbed-os-example-sockets/issues/197#issuecomment-930266687

jeromecoutant, 2021-10-01:
@Patater

HelloHungryImDad, 2022-10-31:
I am facing the same issue with Mbed Studio 1.4.4 and the following tool versions for mbed-os-blinky with mbed-os 6.13.0. 

[image]

Has there been any news on this issue at all?

HelloHungryImDad, 2022-10-31:
This exact process works with Nucleo L432KC, Nucleo F767ZI, and on Disco L475VG-IOT01A. Could it be the CMSIS pack that isn't right?

HelloHungryImDad, 2022-11-01:
Running pyocd with `-vv` flags yields the following additional information:

```
0000587 D STLink probe 0037001A3756501620303658 firmware version: V3J10M3 [stlink]
0000588 D Project directory: /home/rikki/Downloads/Keil.STM32WLxx_DFP.1.1.0 [session]
0000828 D flash algo: [code=0x254] [b1=0x260,0x660] [b2=0x660,0xa60] [stack=0x2a60; 0x2000 b] (ram=0x20000000, 0x8000 b) [flash_algo]
0000831 I Target type is stm32wl55jcix [board]
0000834 D STLink probe 0037001A3756501620303658 firmware version: V3J10M3 [stlink]
0000834 D actual SWD frequency is 1000 kHz [stlink]
0000834 D Running task load_svd [sequencer]
0000891 D Running task pre_connect [sequencer]
0000897 I Asserting reset prior to connect [coresight_target]
0000908 D Running task dp_init [sequencer]
0000913 D Running task lock_probe [sequencer]
0000919 D Running task get_probe_capabilities [sequencer]
0000925 D Running task connect [sequencer]
0000947 D Default wire protocol selected; using SWD [dap]
0000953 I DP IDR = 0x6ba02477 (v2 rev6) [dap]
0000953 D Running task clear_sticky_err [sequencer]
0000953 D Running task power_up_debug [sequencer]
0000954 D Running task check_version [sequencer]
0000954 D Running task unlock_probe [sequencer]
0000954 D Running task create_discoverer [sequencer]
0000954 D Running task discovery [sequencer]
0000954 D Running task find_aps [sequencer]
0000956 D Running task create_aps [sequencer]
0000956 D Running task create_ap.0 [sequencer]
0000957 D Using accelerated memory access interface for AHB-AP#0 [ap]
0000957 I AHB-AP#0 IDR = 0x24770011 (AHB-AP var1 rev2) [ap]
0000958 D AHB-AP#0 default HPROT=3 HNONSEC=0 [ap]
0000959 D AHB-AP#0 implemented HPROT=3 HNONSEC=0 [ap]
0000960 D Running task create_ap.1 [sequencer]
0000960 D Using accelerated memory access interface for AHB-AP#1 [ap]
0000960 I AHB-AP#1 IDR = 0x84770001 (AHB-AP var0 rev8) [ap]
0000961 D AHB-AP#1 default HPROT=3 HNONSEC=1 [ap]
0000962 D AHB-AP#1 implemented HPROT=f HNONSEC=1 [ap]
0000963 D Running task find_components [sequencer]
0000963 D Running task init_ap.0 [sequencer]
0000967 I AHB-AP#0 Class 0x1 ROM table #0 @ 0xe00ff000 (designer=020:ST part=497) [rom_table]
0000970 I [0]<e000e000:SCS v7-M class=14 designer=43b:Arm part=000> [rom_table]
0000971 I [1]<e0001000:DWT v7-M class=14 designer=43b:Arm part=002> [rom_table]
0000972 I [2]<e0002000:FPB v7-M class=14 designer=43b:Arm part=003> [rom_table]
0000973 I [3]<e0000000:ITM v7-M class=14 designer=43b:Arm part=001> [rom_table]
0000975 I [4]<e0040000:TPIU M4 class=9 designer=43b:Arm part=9a1 devtype=11 archid=0000 devid=ca0:0:0> [rom_table]
0000975 D [5]<fff42002 not present> [rom_table]
0000977 I [6]<e0043000:CTI class=9 designer=43b:Arm part=906 devtype=14 archid=0000 devid=40800:0:0> [rom_table]
0000977 D Running task init_ap.1 [sequencer]
0000979 E Transfer error while reading AHB-AP#1 ROM table: Memory transfer fault (read) @ 0xf0000fd0-0xf000102f [ap]
Traceback (most recent call last):
  File "/home/rikki/.local/lib/python3.10/site-packages/pyocd/coresight/ap.py", line 797, in find_components
    cmpid.read_id_registers()
  File "/home/rikki/.local/lib/python3.10/site-packages/pyocd/coresight/rom_table.py", line 124, in read_id_registers
    regs = self.ap.read_memory_block32(self.top_address + self.IDR_READ_START, self.IDR_READ_COUNT)
  File "/home/rikki/.local/lib/python3.10/site-packages/pyocd/utility/concurrency.py", line 29, in _locking
    return func(self, *args, **kwargs)
  File "/home/rikki/.local/lib/python3.10/site-packages/pyocd/coresight/ap.py", line 1216, in _accelerated_read_memory_block32
    return self._accelerated_memory_interface.read_memory_block32(addr, size,
  File "/home/rikki/.local/lib/python3.10/site-packages/pyocd/probe/stlink_probe.py", line 301, in read_memory_block32
    return conversion.byte_list_to_u32le_list(self._link.read_mem32(addr, size * 4, self._apsel, csw))
  File "/home/rikki/.local/lib/python3.10/site-packages/pyocd/probe/stlink/stlink.py", line 473, in read_mem32
    return self._read_mem(addr, size, Commands.JTAG_READMEM_32BIT, self.MAXIMUM_TRANSFER_SIZE, apsel, csw)
  File "/home/rikki/.local/lib/python3.10/site-packages/pyocd/probe/stlink/stlink.py", line 423, in _read_mem
    raise exc
pyocd.core.exceptions.TransferFaultError: Memory transfer fault (read) @ 0xf0000fd0-0xf000102f
0000981 D Running task create_cores [sequencer]
0000981 D Creating SCS component [discovery]
0000982 D selected core #0 [soc_target]
0000984 I CPU core #0 is Cortex-M4 r0p1 [cortex_m]
0000985 D Running task set_default_reset_type [sequencer]
0000985 D Running task create_components [sequencer]
[... 143 lines cut ...]
[==============================================    ]  93%0002005 D resuming core 0 [cortex_m]
0002005 D added=[] removed=[] [manager]
0002005 D bps after flush={} [manager]
[================================================  ]  96%0002027 D resuming core 0 [cortex_m]
0002027 D added=[] removed=[] [manager]
0002027 D bps after flush={} [manager]
[================================================= ] 100%0002036 D resuming core 0 [cortex_m]
0002036 D added=[] removed=[] [manager]
0002036 D bps after flush={} [manager]
[==================================================] 100%
0002041 D Estimated sector erase programmed page count: 23 [builder]
0002041 D Actual sector erase programmed page count: 23 [builder]
0002041 D set reset catch, core 0 [cortex_m]
0002041 D halting core 0 [cortex_m]
0002042 D reset, core 0, type=SW_SYSRESETREQ [cortex_m]
[... cut ...] [cortex_m]
0002160 I Erased 24576 bytes (12 sectors), programmed 23552 bytes (23 pages), skipped 0 bytes (0 pages) at 19.91 kB/s [loader]
0002160 D uninit session <pyocd.core.session.Session object at 0x7f838d2e3310> [session]
0002161 D uninit board <pyocd.board.mbed_board.MbedBoard object at 0x7f838d2e2830> [board]
0002162 D resuming core 0 [cortex_m]
0002162 D added=[] removed=[] [manager]
0002162 D bps after flush={} [manager]
```

It's definitely not happy about AHB-AP#1 whatever that means.

jeromecoutant, 2022-11-07:
> I am facing the same issue with Mbed Studio 1.4.4 and the following tool versions for mbed-os-blinky with mbed-os 6.13.0.

Why mbed-os 6.13 and not 6.16 ?

hallard, 2022-11-07:
@jeromecoutant you mean that this issue should be fixed in 6.16?

jeromecoutant, 2022-11-07:
no, but I don't remember what 6.13 contains....