ARMmbed/mbed-os #15047: CMSIS defines are too generic

ladislas, 2021-09-02:
### Description of defect

Some TARGET/CMSIS defines are given names that are really generic such as `DMA2D` in https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32F7/STM32Cube_FW/CMSIS/stm32f769xx.h

This prevents the user from defining variables with the same name. In our case we want to have:

```cpp
namespace interface {

class DMA2D
{
	// ...
};

}
```

To define the DMA2D interface other components would need.

I could rename it DMA2DBase but this breaks our naming rules.

DMA2D is not the only one and the same issues could arise with `JPEG`, `DSI`, `DAC`, `ADC`, etc.

#### Target(s) affected by this defect ?

SM32F769 but others might be as well.

#### Toolchain(s) (name and version) displaying this defect ?

arm-none-eabi-c++ (GNU Arm Embedded Toolchain 10-2020-q4-major) 10.2.1 20201103 (release)

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.14.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

latest

#### How is this defect reproduced ? 

See description on how to reproduce.

jeromecoutant, 2021-09-02:
Hi
Please open an issue in the source repo:
https://github.com/STMicroelectronics/cmsis_device_f7

0xc0170, 2021-09-02:
I wonder what naming scheme should there. Everyone just follows their reference manual when creating `Peripheral_declaration`. I don't think CMSIS dictates anything towards this direction.

One way would be to not allow this MCU header get leaked outside of Mbed OS library (only if anything outside requests it and includes it) or?

ladislas, 2021-09-02:
@jeromecoutant done! thanks :) https://github.com/STMicroelectronics/cmsis_device_f7/issues/2

ladislas, 2021-09-18:
@jeromecoutant sadly the issue in ST's repo was closed for fair reasons.

It seems that you don't just pull the file from there to add to mbed-os, but you do some modifications before.

Would it be possible to add something like this:

```cpp
/** @addtogroup Peripheral_declaration
  * @{
  */
#ifndef HIDE_PERIPHERAL_DECLARATION

#define TIM2                ((TIM_TypeDef *) TIM2_BASE)
#define TIM3                ((TIM_TypeDef *) TIM3_BASE)
// ...
#define DMA2D               ((DMA2D_TypeDef *)DMA2D_BASE)
// ...
#define JPEG                ((JPEG_TypeDef *) JPEG_BASE)
#define DSI                 ((DSI_TypeDef *)DSI_BASE)

#endif // HIDE_PERIPHERAL_DECLARATION
/**
  * @}
  */
```

This way it won't break anything in current code and people who need it can just define `HIDE_PERIPHERAL_DECLARATION` somewhere.

If it works for you I can make a PR :)

jeromecoutant, 2021-09-20:
Hi
Issue is to find an easy way to keep the patch when we update drivers files for each ST Cube FW...
I tried to remove these peripheral declarations, but it seems that lotsv of HAL ST files are based on this... :-(

Ex: https://github.com/STMicroelectronics/STM32CubeF7/blob/master/Drivers/STM32F7xx_HAL_Driver/Inc/stm32f7xx_hal_dac.h#L35
`#if defined(DAC)`

ladislas, 2021-09-20:
Ah yes I see... well then I'm not sure there's an easy solution :(

0xc0170, 2021-10-04:
I am inclined to close this issue. There's no easy solution and we won't be able to fix all implementations out there.

Closing as won't fix.

closed 2021-10-04 by 0xc0170