ARMmbed/mbed-os #15053: mbed-tools fails build on first run and with "--clean"

kalle16lab, 2021-09-06:
### Description of defect

Doing  `mbed-tools compile -m <custom target> -t GCC_ARM --clean` always fails with 

```
_deps/googletest-src/googletest/include/gtest/internal/gtest-port.h:2039:40: error: 'fileno' was not declared in this scope; did you mean 'file'?
 2039 | inline int FileNo(FILE* file) { return fileno(file); }
      |                                        ^~~~~~
      |                                        file
```
Doing `mbed-tools compile -m <custom target> -t GCC_ARM` fails on first run but completes without issue on second run.

This seems to be an issue with googletest requiring CXXFLAGS=-std=gnu++14 (see: https://github.com/google/googletest/issues/813)

Doing a verbose build I can confirm that googletest is being currently built with CXXFLAGS=-std=c++14.

```
/opt/mbed-os-toolchain/gcc-arm-none-eabi-9-2019-q4-major/bin/arm-none-eabi-g++ -DUNITTEST -isystem _deps/googletest-src/googletest/include -isystem _deps/googletest-src/googletest -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -fmessage-length=0 -fno-exceptions -ffunction-sections -fdata-sections -funsigned-char -fomit-frame-pointer -g3 -mthumb -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=softfp -Wall -Wshadow -Wno-error=dangling-else -DGTEST_HAS_PTHREAD=0 -fexceptions -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -std=c++14 -MD -MT _deps/googletest-build/googletest/CMakeFiles/gtest_main.dir/src/gtest_main.cc.obj -MF _deps/googletest-build/googletest/CMakeFiles/gtest_main.dir/src/gtest_main.cc.obj.d -o _deps/googletest-build/googletest/CMakeFiles/gtest_main.dir/src/gtest_main.cc.obj -c _deps/googletest-src/googletest/src/gtest_main.cc
```

Is there any way to exclude googletest from the build?

#### Target(s) affected by this defect ?
N/A

#### Toolchain(s) (name and version) displaying this defect ?
gcc-arm-none-eabi-9-2019-q4-major

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.14.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-tools 7.32.0

#### How is this defect reproduced ? 
Currently this is happening on a proprietary BLE project I can't share and curiously using the same custom target to build BLE examples do not incur this behaviour.

0xc0170, 2021-09-06:
As you stated, the error does not happen for a normal example. how can we reproduce the problem?

> mbed-tools compile -m <custom target> -t GCC_ARM --clean

This should not build the test. The same issue persist on master as well?

From the command you shared, looks like you get into this condition https://github.com/ARMmbed/mbed-os/blob/master/CMakeLists.txt#L36 (verify the two flags if they are set - building testing and building greentea tests).

noonfom, 2021-09-06:
The issue appears to be with mbed-tools 7.32.0; solved by rolling back to 7.31.0. Only difference is one [commit](https://github.com/ARMmbed/mbed-tools/commit/ab5e7bf371c686654ec8aca5afd008d742525852) 😉

kalle16lab, 2021-09-07:
this is the contents of `mbed_config.cmake` after doing a `mbed-tools configure ...`
```
# Copyright (C) 2020 Arm Mbed. All rights reserved.
# SPDX-License-Identifier: Apache-2.0

# Automatically generated configuration file.
# DO NOT EDIT. Content may be overwritten.

set(MBED_TOOLCHAIN "GCC_ARM" CACHE STRING "")
set(MBED_TARGET "<our target>" CACHE STRING "")
set(MBED_CPU_CORE "Cortex-M4F" CACHE STRING "")
set(MBED_C_LIB "std" CACHE STRING "")
set(MBED_PRINTF_LIB "minimal-printf" CACHE STRING "")
set(MBED_OUTPUT_EXT "" CACHE STRING "")
set(MBED_GREENTEA_TEST_RESET_TIMEOUT "" CACHE STRING "")

list(APPEND MBED_TARGET_SUPPORTED_C_LIBS 
    std
    small
)

list(APPEND MBED_TARGET_SUPPORTED_APPLICATION_PROFILES 
    full
    bare-metal
)

list(APPEND MBED_TARGET_LABELS
    FAMILY_Apollo3
    AMA3B1KK
    M4
    CORTEX
    <our custom target>
    LIKE_CORTEX_M4
    CORTEX_M
    Target
    RTOS_M4_M7

    CORDIO
    Apollo3
    CORDIO_LL
    Ambiq_Micro

    FLASHIAP

    BLE
)

# target
set(MBED_TARGET_DEFINITIONS
    COMPONENT_FLASHIAP=1

    FEATURE_BLE=1

    DEVICE_SLEEP=1
    DEVICE_SERIAL=1
    DEVICE_LPTICKER=1
    DEVICE_USTICKER=1
    DEVICE_FLASH=1
    DEVICE_INTERRUPTIN=1
    DEVICE_SPI=1
    DEVICE_STDIO_MESSAGES=1
    DEVICE_MPU=1
    DEVICE_I2C=1

    TARGET_FAMILY_Apollo3
    TARGET_AMA3B1KK
    TARGET_M4
    TARGET_CORTEX
    <our custom target>
    TARGET_LIKE_CORTEX_M4
[... 465 lines cut ...]
# If you do want to change a value, simply edit, save, and exit the editor.
# The syntax for the file is as follows:
# KEY:TYPE=VALUE
# KEY is the name of a variable in the cache.
# TYPE is a hint to GUIs for the type of VALUE, DO NOT EDIT TYPE!.
# VALUE is the current value for the KEY.

########################
# EXTERNAL cache entries
########################

//Enable building the examples
BUILD_EXAMPLES:BOOL=ON

//Builds the googlemock subproject
BUILD_GMOCK:BOOL=ON

//Build greentea tests only.
BUILD_GREENTEA_TESTS:BOOL=OFF

//Build the testing tree.
BUILD_TESTING:BOOL=ON
```

I don't understand where these external cache entries are coming from or how to override them. Maybe some mbed-os internal library I'm using has building tests hard-coded?

kalle16lab, 2021-09-07:
Update:
 This seems to be a problem with the Ninja build tool. I have confirmed that doing the following steps:

`mbed-tools configure -m <...> -t GCC_ARM`
`cmake -S . -B cmake_build/<...>/develop/GCC_ARM`
and finally
`cmake --build cmake_build/<...>/develop/GCC_ARM`

does build without errors but doing the same step wise build with Ninja fails the same way as doing a `mbed-tools compile` does

Seems like for now using cmake to build is a workaround.

kalle16lab, 2021-09-09:
I started out from a clean master and added back our target. The issue persisted but commenting out https://github.com/ARMmbed/mbed-os/blob/82036c85dfc7186347598b1def4a3688c64a60a9/CMakeLists.txt#L34 fixes the issue. 

Why is CTest included by default? Shouldn't it also be only included when testing is enabled?

0xc0170, 2021-09-10:
Please do the same as this commit https://github.com/ARMmbed/mbed-os-example-sockets/commit/56f3cc7b2f27b7e7b53ddac918987c596be41909 . It should fix the issue.

0xc0170, 2021-09-27:
I'll close this as resolved.

closed 2021-09-27 by 0xc0170