ARMmbed/mbed-os #15054: STM32WL change gpio mode in Release take 8 times longer than in Debug/Develop

hallard, 2021-09-06:
### Description of defect

On STM32WL board (Nucleo WL55JC)  and may be other)

#### Target(s) affected by this defect ?

NUCLEO_WL55JC and STM32WL in `customtargets.json`

#### Toolchain(s) (name and version) displaying this defect ?

GNU Arm Embedded Toolchain 10-2020-q4-major 10.2.1 20201103 (release)

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-99.99.99
    
#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed studio V1.4.1 
mbed-tools 7.27.0

#### How is this defect reproduced ? 

Compile and run following code

```cpp
int main()
{
    DigitalInOut gpio(PB_2);
    Timer timer;

    printf("--Starting--\r\n");
    uint16_t dly1; 
    uint16_t dly2;

    gpio.output();
    gpio.mode(OpenDrain);
    timer.start();
    gpio.input();
    dly1 = timer.elapsed_time().count();
    timer.reset();
    gpio.read();
    timer.stop();
    dly2 = timer.elapsed_time().count();
    printf("delay1 = %dus\r\n", dly1);
    printf("delay2 = %dus\r\n", dly2);
}
```

Debug (and Develop) shows 
`mbed-tools compile --profile Debug  -m NUCLEO_WL55JC -t GCC_ARM --flash`
```
--Starting--
delay1 = 13us
delay2 = 10us
```

Release shows
`mbed-tools compile --profile Release  -m NUCLEO_WL55JC -t GCC_ARM --flash`
```
--Starting--
delay1 = 110us
delay2 = 87us
```

I need to use a DS18B20 (customer choice) and each time I'm using this (each target) it's a pain with timing. 
Any idea are welcome

mbedmain, 2021-09-06:
@hallard thank you for raising this issue.Please take a look at the following comments:

Could you add some more detail to the description? A good description should be at least 25 words.

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

jeromecoutant, 2021-09-07:
Values confirmed with WL and ARM:
```
--Starting-- ==> debug
delay1 = 14us
delay2 = 11us
--Starting-- ==> develop
delay1 = 11us
delay2 = 8us
--Starting-- ==> release
delay1 = 110us
delay2 = 86us
```

Tried the same appli with DISCO L475. Seems value is the same for all profiles:
```
--Starting--
delay1 = 7us
delay2 = 5us
```

Here is diff...
```
$ diff tools/profiles/debug.json tools/profiles/release.json
8,12c8,11
<                    "-fomit-frame-pointer", "-Og", "-DMBED_DEBUG",
<                    "-DMBED_TRAP_ERRORS_ENABLED=1"],
<         "asm": ["-c", "-g3", "-x", "assembler-with-cpp"],
<         "c": ["-c", "-g3", "-std=gnu11"],
<         "cxx": ["-c", "-g3", "-std=gnu++14", "-fno-rtti", "-Wvla"],
---
>                    "-fomit-frame-pointer", "-Os", "-DNDEBUG", "-g"],
>         "asm": ["-c", "-x", "assembler-with-cpp"],
>         "c": ["-c", "-std=gnu11"],
>         "cxx": ["-c", "-std=gnu++14", "-fno-rtti", "-Wvla"],
```

hallard, 2021-09-07:
Yeah some STM32 targets does have longer time to switch gpio mode (looks like STM32WL is one of them) 
Check ou this [post](https://forums.mbed.com/t/ds1820-onewire-library-doesnt-work-on-mbedos5-mbedos6-nucleo-f303re/9937/) about explanation.
But that does not explain why it's so longer in release mode

jeromecoutant, 2021-09-07:
Adding 2 gpio to check with analyser:
```
    test1 = 0;
    test2 = 0;
    gpio.output();
    gpio.mode(OpenDrain);
    test2 = 1;
    timer.start();
    test1 = 1;
    gpio.input();
    test1 = 0;
    dly1 = timer.elapsed_time().count();
    test2 = 0;
    timer.reset();
    test1 = 1;
    gpio.read();
    test1 = 0;
    timer.stop();
    test2 = 1;
    dly2 = timer.elapsed_time().count();
````
test1 was up 24.6us and 1.2us
test2 
[image]

hallard, 2021-09-07:
You beat me, was in my todo but need to flash/provision some devices for customer deployment before going back to this point.
Looking you graph, switching gpio from output to input takes 22us (in Debug I assume?)

jeromecoutant, 2021-09-07:
I am working in release profile, gpio.input() takes 24.6us

hallard, 2021-11-28:
just wondering is that issue is still active or fixed?

jeromecoutant, 2021-11-29:
I don't think something has changed

caproc, 2022-02-16:
I'm having the same issue with my STM32WL. Can't use DS18B20 sensor unless using UART instead of single gpio. I'm also getting different values from ADC (DigitalIn) in Release and Debug mode when reading battery voltage (through voltage divider). Debug mode gives correct value, but Release mode lot less. E.g. actual battery voltage of 3.6V results close to 2V in Release mode. Any idea what might cause these differences in Release vs Debug build profiles?

caproc, 2022-02-16:
Sorry, meant AnalogIn and not DigitalIn.. :)