ARMmbed/mbed-os #15064: Nucleo-F031K6 fails to work with mbed-os 5

maxgerhardt, 2021-09-12:
### Description of defect

When following https://os.mbed.com/docs/mbed-studio/current/getting-started/index.html and changing the mbed-os version to 5.15.7 and selecting the NUCLEO-F031K6 target, the mbed-os-example-blinky-baremetal project (or a completely empty main code) fails to compile due to not enough RAM.

```
Link: mbed-os-example-blinky-baremetal
[Warning] @0,0: L3912W: Option 'legacyalign' is deprecated.
[Error] @0,0: L6221E: Execution region RW_IRAM1 with Execution range [0x200000b0,0x2000031c) overlaps with Execution region ARM_LIB_STACK with Execution range [0x20000000,0x20001000).
[Error] @0,0: L6221E: Execution region ARM_LIB_HEAP with Execution range [0x20000320,0x20000320) overlaps with Execution region ARM_LIB_STACK with Execution range [0x20000000,0x20001000).
Warning: L3912W: Option 'legacyalign' is deprecated.
Error: L6221E: Execution region RW_IRAM1 with Execution range [0x200000b0,0x2000031c) overlaps with Execution region ARM_LIB_STACK with Execution range [0x20000000,0x20001000).
Error: L6221E: Execution region ARM_LIB_HEAP with Execution range [0x20000320,0x20000320) overlaps with Execution region ARM_LIB_STACK with Execution range [0x20000000,0x20001000).
Finished: 0 information, 1 warning and 2 error messages.
[ERROR] Warning: L3912W: Option 'legacyalign' is deprecated.
Error: L6221E: Execution region RW_IRAM1 with Execution range [0x200000b0,0x2000031c) overlaps with Execution region ARM_LIB_STACK with Execution range [0x20000000,0x20001000).
Error: L6221E: Execution region ARM_LIB_HEAP with Execution range [0x20000320,0x20000320) overlaps with Execution region ARM_LIB_STACK with Execution range [0x20000000,0x20001000).
Finished: 0 information, 1 warning and 2 error messages.
```

About the same happens with mbed-os 5.15.6 + PlatformIO (+ GCC)

```
Linking .pio\build\nucleo_f031k6\firmware.elf
[..].pio/build/nucleo_f031k6/STM32F031X6.ld.link_script.ld:82 cannot move location counter backwards (from 0000000020000368 to 0000000020000000)
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\nucleo_f031k6\firmware.elf] Error 1
```

Yet you list the target as supported.

https://github.com/ARMmbed/mbed-os/blob/0be7685a27f28c02737f42055f4a9c182a7b483c/targets/targets.json#L2204-L2206

#### Target(s) affected by this defect ?

`NUCLEO_F031K6`

#### Toolchain(s) (name and version) displaying this defect ?

ARMC6.

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os 5.15.7

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Mbed studio

#### How is this defect reproduced ? 

1. Create new MBed studio project according to https://os.mbed.com/docs/mbed-studio/current/getting-started/index.html 
2. Select NUCLEO-F031K6 as target
3. Select mbed-os 5.15.7 as the mbed-os version
[image]
4. Change code to bare minimum
```cpp
#include "mbed.h"

int main()
{
    while (true)
    {
    }
}
```
5. Build the project and observe error

jeromecoutant, 2021-09-13:
Hi

Yes, mbed can not support MCU with only 4K RAM....

maxgerhardt, 2021-09-13:
Why is it in the `targets.json` of mbed-os 5 then 😄

Also at https://os.mbed.com/platforms/ST-Nucleo-F031K6/

[image]

jeromecoutant, 2021-09-16:
Hi
Please add in targets.json file for NUCLEO_F031K6:
`"overrides": {"boot-stack-size"   : "0x400"},`

Or build with `-DMBED_BOOT_STACK_SIZE=0x400`

jeromecoutant, 2021-09-16:
FYI: issue about boot-stack-size was raised in #10319 and corrected in mbed-os-6.3.0 with #13452

MarceloSalazar, 2021-09-16:
Closing as considered fixed.

closed 2021-09-16 by MarceloSalazar