ARMmbed/mbed-os #15069: I2C_SDA and I2C_SCL removed from NUCLEO_L432KC

ianhattendorf, 2021-09-15:
### Description of defect

I2C pin names were removed in #14480

#### Target(s) affected by this defect ?
NUCLEO_L432KC

#### Toolchain(s) (name and version) displaying this defect ?
GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.14.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli 1.10.5

#### How is this defect reproduced ? 

Attempt to use I2C_SDA or I2C_SCL when targeting NUCLEO_L432KC

mbedmain, 2021-09-15:
@ianhattendorf thank you for raising this issue.Please take a look at the following comments:

Could you add some more detail to the description? A good description should be at least 25 words.

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

jeromecoutant, 2021-09-15:
Thx @ianhattendorf for issue report

Please check #15074

ianhattendorf, 2021-09-15:
@jeromecoutant that resolves the issue, thanks!

PR #15074 (2021-09-29): STM32 : default SPI and I2C for targets with ARDUINO-NANO
### Summary of changes 

Fix #15069 

Default SPI and I2C pins for arduino nano connectors were wrongly removed.

#### Impact of changes 

#### Migration actions required 

### Documentation 

----------------------------------------------------------------------------------------------------------------
### Pull request type
closed 2021-09-29 by 0xc0170