ARMmbed/mbed-os #15089: Build scripts using deprecated `--legacyalign`  build option

MarceloSalazar, 2021-09-20:
### Description of defect

Mbed OS is using `--legacyalign` option in build scripts:
https://github.com/ARMmbed/mbed-os/search?q=legacyalign

This is deprecated since Arm Compiler 6.6
https://developer.arm.com/tools-and-software/embedded/arm-compiler/downloads/version-6/6-6

This is causing all builds to display an unnecessary warning message:
[image]

Issue reported by @ARMmbed/team-maximintegrated 

#### Target(s) affected by this defect ?

All

#### Toolchain(s) (name and version) displaying this defect ?

Arm Compiler 6.16

#### What version of Mbed-os are you using (tag or sha) ?

4cfbea43cabe86bc3ed7a5287cd464be7a218938

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed cli 1.10.4

#### How is this defect reproduced ? 

Compile any Mbed OS application using Mbed CLI.

jeromecoutant, 2021-09-20:
duplication of #14290 ?

MarceloSalazar, 2021-09-22:
Closing as considered a duplicate.

closed 2021-09-22 by MarceloSalazar