ARMmbed/mbed-os #15106: TLSSocket::listen() is not supported

blackwiz4rd, 2021-09-24:
### Description of defect

Hi, I would like to understand the purpose of the TLSSocket and the difference between using TLSSocket and using mbedtls_* (as in this example https://github.com/ARMmbed/mbed-os-example-tls/blob/master/tls-client/HelloHttpsClient.cpp) functions.
In addition, I would like to point out that TLSSocket::listen() is not yet implemented (https://os.mbed.com/docs/mbed-os/v6.14/apis/tlssocket.html) as it returns "NSAPI_ERROR_UNSUPPORTED" (https://github.com/ARMmbed/TLSSocket/blob/master/TLSSocketWrapper.cpp). If possible I would like to understand the reason behind this method not being implemented.
Best regards.

#### How is this defect reproduced ? 
Use TLSSocket::listen() function.

mbedmain, 2021-09-24:
@blackwiz4rd thank you for raising this issue.Please take a look at the following comments:

What target(s) are you using?
What toolchain(s) are you using?
What Mbed OS version are you using?
It would help if you could also specify the versions of any tools you are using?
 
NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

0xc0170, 2021-09-28:
cc @ARMmbed/mbed-os-connectivity