ARMmbed/mbed-os #15108: DISCO_F769NI has QSPI timeouts

Jookia, 2021-09-27:
### Description of defect

When reading from a LittleFSv2 filesystem, reads will fail sometimes.
Adding tracing or block device slicing causes this issue to become less pronounced.

Here's an example error with tracing (and a printf in HAL_QSPI_* functions):

```
[2317.075817 0.000972] ÿClosing "/fs/numbers.txt"... OK
[2317.078827 0.003010] Opening the root directory... OK
[2317.081825 0.002998] root directory:
[2317.082858 0.001033]     .
[2317.083829 0.000971]     ..
[2322.085432 5.001603] [ERR ][STQS]: HAL_QSPI_Receive error 1 status 1
[2322.090482 0.005050] [ERR ][QSPIF]: QSPI Read failed
[2322.094494 0.004012] [ERR ][QSPIF]: Read Command failed
[2322.099857 0.005363] Closing the root directory... OK
[2322.101475 0.001618] Opening "/fs/numbers.txt"... [ERR ][STQS]: HAL_QSPI_Receive error 1 status 1
[2322.109472 0.007998] [ERR ][QSPIF]: QSPI Read failed
[2322.113477 0.004004] [ERR ][QSPIF]: Read Command failed
[2322.118441 0.004965] Fail :(
[2322.119431 0.000990]
[2322.119441 0.000010]
[2322.119447 0.000006] ++ MbedOS Error Info ++
[2322.121459 0.002012] Error Status: 0x80FF0100 Code: 256 Module: 255
[2322.125466 0.004007] Error Message: Fatal Run-time error
[2322.128478 0.003012] Location: 0x8000415
[2322.130466 0.001988] Error Value: 0x0
[2322.132442 0.001976] Current Thread: main Id: 0x20002854 Entry: 0x8002951 StackSize: 0x1000 StackMem: 0x20001830 SP: 0x2000277C
[2322.141476 0.009034] For more info, visit: https://mbed.com/s/error?error=0x80FF0100&tgt=DISCO_F769NI
[2322.148481 0.007005] -- MbedOS Error Info --
[2322.151437 0.002956] error:  (-4001)
[2322.152460 0.001023]
```

(times are added by grabserial)

#### Target(s) affected by this defect ?

32F769IDISCOVERY

#### Toolchain(s) (name and version) displaying this defect ?

arm-none-eabi- 11.2.0 from Arch Linux
arm-none-eabi- 10.2.1 from Arch Linux User Repository
mbed-cli installed from pip

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.14.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed-cli from pip, so 1.10.5
See toolchains reported earlier

#### How is this defect reproduced ? 

Run the LittleFS example with the filesystem changed to LittleFSv2. Keep rebooting the board until failure.

The testing setup we use is this:

[main.cpp](https://github.com/ARMmbed/mbed-os/files/7233465/main.cpp.txt)
[target.json](https://github.com/ARMmbed/mbed-os/files/7233470/target.json.txt)
[hal_log.patch](https://github.com/ARMmbed/mbed-os/files/7233469/hal_log.patch.txt)

This bug seems fairly similiar to https://github.com/ARMmbed/mbed-os/issues/10049

Re-applying the changes to https://github.com/ARMmbed/mbed-os/commit/5c872a4c0669f9d2ef70a9401bbaa469b9881f57 as [mbed_fix.patch](https://github.com/ARMmbed/mbed-os/files/7233491/mbed_fix.patch.txt) seems to fix the issue. Maybe the HAL needs to define QSPI1_V1_0 for the 32F769NI? Or maybe it's a false positive.

jeromecoutant, 2021-09-27:
Hi

> Maybe the HAL needs to define QSPI1_V1_0 for the 32F769NI? Or maybe it's a false positive.

Could you try to remove the `#if  defined(QSPI1_V1_0)`
in
https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32F7/STM32Cube_FW/STM32F7xx_HAL_Driver/stm32f7xx_hal_qspi.c#L1024
and/or
https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32F7/STM32Cube_FW/STM32F7xx_HAL_Driver/stm32f7xx_hal_qspi.c#L1115

To be sure ?

Thx

Jookia, 2021-09-27:
Hi, yes my patch mbed_fix.patch does that and seems to fix the issue or make it harder to reproduce.

jeromecoutant, 2021-09-27:
But both remove are needed, or only 1 ?

Jookia, 2021-09-27:
I'm not sure, I removed all of them I could see.
Should I try seeing if only one is needed?

Jookia, 2021-09-28:
Uncommenting EITHER of those seems to fix the issue.

0xc0170, 2022-07-22:
As it's been almost a year without any update, I'll close this as won't fix.

closed 2022-07-22 by 0xc0170