ARMmbed/mbed-os #15122: STM32F303 CAN tderror() gives always zero

pilotak, 2021-09-30:
### Description of defect

In order to determine if somebody listens on the CAN bus there is a function `tderror()` which is exactly for this purpose, nice article [here](https://www.kvaser.com/about-can/the-can-protocol/can-error-handling/).
The problem is that on STM32F303RE this counter always gives zero even when the can bus is disconnected. I know this mechanism works because i run the same code for the STM32F091RC and it works as expected-ich, why? because if i understad correctly the `tderror` counter should start from zero and with next error it should add another 8 points, but on the STM32F091 it starts with 128 (Error passive) straight away and then adds another 8. But i can put this into another issue so we don't mix this up with F303 (if this is actually an issue)

F091 log:
```
tderr_prev:0, tderr: 0, sent: 1
tderr_prev:0, tderr: 0, sent: 1
--- bus disconnected here ---
tderr_prev:0, tderr: 128, sent: 1
Error Passive
tderr_prev:128, tderr: 128, sent: 1
tderr_prev:128, tderr: 128, sent: 0
tderr_prev:128, tderr: 128, sent: 0
tderr_prev:128, tderr: 128, sent: 0
tderr_prev:128, tderr: 128, sent: 0
--- bus reconnected here ---
tderr_prev:128, tderr: 133, sent: 1
tderr_prev:133, tderr: 132, sent: 1
Bus recovered
tderr_prev:132, tderr: 131, sent: 1
tderr_prev:131, tderr: 130, sent: 1
tderr_prev:130, tderr: 129, sent: 1
tderr_prev:129, tderr: 128, sent: 1
tderr_prev:128, tderr: 127, sent: 1
tderr_prev:127, tderr: 126, sent: 1
```

F303 log
```
tderr_prev:0, tderr: 0, sent: 1
tderr_prev:0, tderr: 0, sent: 1
tderr_prev:0, tderr: 0, sent: 1
--- bus disconnected here ---
tderr_prev:0, tderr: 0, sent: 1
tderr_prev:0, tderr: 0, sent: 1
tderr_prev:0, tderr: 0, sent: 1
tderr_prev:0, tderr: 0, sent: 0
tderr_prev:0, tderr: 0, sent: 0
tderr_prev:0, tderr: 0, sent: 0
tderr_prev:0, tderr: 0, sent: 0
tderr_prev:0, tderr: 0, sent: 0
--- bus reconnected here ---
tderr_prev:0, tderr: 0, sent: 1
tderr_prev:0, tderr: 0, sent: 1
```

Please see the code below

#### Target(s) affected by this defect ?
STM32F3 family

#### Toolchain(s) (name and version) displaying this defect ?
GCC ARM 9.3.1

#### What version of Mbed-os are you using (tag or sha) ?

bc01a4e063bd23c31a04448d0a0525b327e102e9

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ? 

```cpp
#include <mbed.h>
RawCAN can(CAN_RXD_pin, CAN_TXD_pin);

int main() {
    can.frequency(250000);
    can.mode(CAN::Normal);

    uint8_t tderr_prev = 0;
    bool bus_state = true;

    while (1) {
        CANMessage msg;
        msg.id = 0;
        msg.id |= 6 << 26; // priority
        msg.id |= 0xFF00 << 8; // SPN
        msg.id |= 0xA0; // SA
        msg.len = 8;
        msg.type = CANData;
        msg.format = CANExtended;
        memset(msg.data, 0, 8);

        int sent = can.write(msg);
        uint8_t tderr = can.tderror();
        printf("tderr_prev:%u, tderr: %u, sent: %i\n", tderr_prev, tderr, sent);

        if (tderr > 127) {
            // (maximum tderror count 255) - (each error is + 8 for counter)
            if (tderr > 247) {
                if (tderr == tderr_prev) {
                    printf("Bus Off\n");
                }

            } else {
                if (tderr >= tderr_prev) {
                    if (bus_state) {
                        bus_state = false;
                        printf("Error Passive\n");
                    }

                } else {
                    if (!bus_state) {
                        bus_state = true;
                        printf("Bus recovered\n");
                    }
                }
            }

        } else if (!bus_state) {
            bus_state = true;
            printf("Bus Active\n");
        }

        tderr_prev = tderr;

        ThisThread::sleep_for(1000ms);
    }
}
```

mbedmain, 2021-09-30:
@pilotak thank you for raising this issue.Please take a look at the following comments:

It would help if you could also specify the versions of any tools you are using?
 
NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

jeromecoutant, 2021-10-01:
I have compared F3 and F0 code, I don't see any difference :-(

pilotak, 2021-10-02:
CAN transceiver used in conjunction with F091 - MCP2562FD-H/MF on pins PB8/9
CAN transceiver used in conjunction with F303 - SN65HVD231 on pins PA11/12

That is the only difference on my side

jeromecoutant, 2021-10-04:
I just get the information form ST CAN expert that there is no difference between F091 and F303... :-(

About chosen pins: PA11/PA12
These are USB pins, so maybe you could change to PB8/PB9 ?

pilotak, 2021-10-04:
well this is going to be difficult as this is on the custom board but i will try my best

pilotak, 2021-11-03:
Did i say Monday? i probably meant Wednesday :grinning: anyway i switched it over to PB8/PB9 still the same result. So i decided to replace the transceiver as this was the only difference. I didn't have the SN65HVD231 so i used SN65HVD230 instead. Still no success.

Than i realised where the problem is. When i disconnect the bus there is no 120ohm termination resistor left on the board (i have them else where on the bus). So adding 120R to the board makes the bus valid (even though wires have no connection on the other side) and the CAN controller fires up the passive error correctly.

I appologize for raising up this issue as the problem was on my side.

closed 2021-11-03 by pilotak