ARMmbed/mbed-os #15125: No add_ignore_patterns neither in class mbedToolchain, nor any inheritor

KKoovalsky, 2021-10-03:
### Description of defect

For the Python build API, if `ignore` parameter, supplied to `prepare_toolchain` is not `None`, then the build API crashes.

#### Target(s) affected by this defect ?

#### Toolchain(s) (name and version) displaying this defect ?

All of them. There is no `add_ignore_patterns()` in `mbedToolchain`, `GCC`, `ARM`, `IAR` defined. Those classes can be found under `tools/toolchains` directory.

#### What version of Mbed-os are you using (tag or sha) ?

`master` branch, SHA: `7f2b6d0a197520225a7cacdafe963226e877ff57`.

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ? 

1. Call python3 build API method:

```
from tools.build_api import prepare_toolchain
# ...
prepare_toolchain( ..., ignores=['some/path'])
```

2. Get an error:

> 'GCC_ARM' object has no attribute 'add_ignore_patterns'

---

The issue is in `build_api.py:497`:

        if ignore:
            toolchain.add_ignore_patterns(root=".", base_path=".", patterns=ignore)

mbedmain, 2021-10-03:
@KKoovalsky thank you for raising this issue.Please take a look at the following comments:

Could you add some more detail to the description? A good description should be at least 25 words.
What target(s) are you using?
It would help if you could also specify the versions of any tools you are using?
 
NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

0xc0170, 2021-10-04:
The old tools were frozen (https://os.mbed.com/blog/entry/Introducing-the-new-Mbed-Tools/).

As this is related to using tools directly (is anywhere in the tools ignore parameter set to not None?), I don't think this could be considered as critical and it shall be closed as won't fix.

closed 2021-10-11 by 0xc0170