ARMmbed/mbed-os #15129: USB mass storage device is broken

Jookia, 2021-10-06:
### Description of defect

Using the USB mass storage device example does not work.
Reading data fails and causes Linux to reset loop the device hoping it will work.

#### Target(s) affected by this defect ?

DISCO_F769NI

#### Toolchain(s) (name and version) displaying this defect ?

GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?

9dd6fb4495655a40645216d7abf2314c7c05f1dc with #15128 applied.

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed-cli 1.10.5
gcc 11.2.0
Linux 5.14.6

#### How is this defect reproduced ? 

Build and run the USB mass storage example.
Wireshark supports looking at Linux USB data, which roughly shows this:

```
No.	Time	Source	Destination	Protocol	Length	Info
31925	492.662852	host	1.5.1	USBMS	95	SCSI: Read(10) LUN: 0x00 (LBA: 0x00000000, Len: 8)
Frame 31925: 95 bytes on wire (760 bits), 95 bytes captured (760 bits) on interface usbmon1, id 0
    Interface id: 0 (usbmon1)
    Encapsulation type: USB packets with Linux header and padding (115)
    Arrival Time: Oct  6, 2021 22:11:10.230344000 AEDT
    [Time shift for this packet: 0.000000000 seconds]
    Epoch Time: 1633518670.230344000 seconds
    [Time delta from previous captured frame: 0.006633000 seconds]
    [Time delta from previous displayed frame: 0.007306000 seconds]
    [Time since reference or first frame: 492.662852000 seconds]
    Frame Number: 31925
    Frame Length: 95 bytes (760 bits)
    Capture Length: 95 bytes (760 bits)
    [Frame is marked: False]
    [Frame is ignored: False]
    [Protocols in frame: usb:usbms]
USB URB
    [Source: host]
    [Destination: 1.5.1]
    URB id: 0xffff98dc1327ef00
    URB type: URB_SUBMIT ('S')
    URB transfer type: URB_BULK (0x03)
    Endpoint: 0x01, Direction: OUT
    Device: 5
    URB bus id: 1
    Device setup request: not relevant ('-')
    Data: present (0)
    URB sec: 1633518670
    URB usec: 230344
    URB status: Operation now in progress (-EINPROGRESS) (-115)
    URB length [bytes]: 31
    Data length [bytes]: 31
    [Response in: 31926]
    [bInterfaceClass: Mass Storage (0x08)]
    Unused Setup Header
    Interval: 0
    Start frame: 0
    Copy of Transfer Flags: 0x00000004, No transfer DMA map
    Number of ISO descriptors: 0
USB Mass Storage
[... 210 lines cut ...]
USB URB
    [Source: 1.5.1]
    [Destination: host]
    URB id: 0xffff98dc1327ef00
    URB type: URB_COMPLETE ('C')
    URB transfer type: URB_BULK (0x03)
    Endpoint: 0x81, Direction: IN
    Device: 5
    URB bus id: 1
    Device setup request: not relevant ('-')
    Data: present (0)
    URB sec: 1633518670
    URB usec: 647318
    URB status: Value too large for defined data type (-EOVERFLOW) (-75)
    URB length [bytes]: 0
    Data length [bytes]: 0
    [Request in: 32251]
    [Time from request: 0.248150000 seconds]
    [bInterfaceClass: Mass Storage (0x08)]
    Unused Setup Header
    Interval: 0
    Start frame: 0
    Copy of Transfer Flags: 0x00000204, No transfer DMA map, Dir IN
    Number of ISO descriptors: 0
```

Jookia, 2021-10-06:
I found something that seems to work:

```
diff --git a/drivers/usb/include/usb/USBMSD.h b/drivers/usb/include/usb/USBMSD.h
index a570df7d82..b516f918c1 100644
--- a/drivers/usb/include/usb/USBMSD.h
+++ b/drivers/usb/include/usb/USBMSD.h
@@ -249,8 +249,8 @@ private:
     // endpoints
     usb_ep_t _bulk_in;
     usb_ep_t _bulk_out;
-    uint8_t _bulk_in_buf[64];
-    uint8_t _bulk_out_buf[64];
+    uint8_t _bulk_in_buf[512];
+    uint8_t _bulk_out_buf[512];
     bool _out_ready;
     bool _in_ready;
     uint32_t _bulk_out_size;
diff --git a/drivers/usb/source/USBDevice.cpp b/drivers/usb/source/USBDevice.cpp
index 6d3cfc87d4..0181561d8c 100644
--- a/drivers/usb/source/USBDevice.cpp
+++ b/drivers/usb/source/USBDevice.cpp
@@ -53,7 +53,7 @@
 #if defined(MAX_PACKET_SIZE_EP0)
 #undef MAX_PACKET_SIZE_EP0
 #endif
-#define MAX_PACKET_SIZE_EP0 64
+#define MAX_PACKET_SIZE_EP0 512
 
 #define USB_MIN(a, b) ((a) > (b) ? (b) : (a))
 
diff --git a/drivers/usb/source/USBMSD.cpp b/drivers/usb/source/USBMSD.cpp
index fcf416c823..c5cde22724 100644
--- a/drivers/usb/source/USBMSD.cpp
+++ b/drivers/usb/source/USBMSD.cpp
@@ -56,7 +56,7 @@
 #define DEFAULT_CONFIGURATION (1)
 
 // max packet size
-#define MAX_PACKET  64
+#define MAX_PACKET  512
 
 // CSW Status
 enum Status {
```

It looks like the HS OTG code is setting up a max packet size of 512 while this code expects 64.

facchinm, 2021-10-07:
Yes, the link was just the tip of the iceberg :slightly_smiling_face: The define is then applied throughout all the other files

0xc0170, 2022-07-22:
As it's been almost a year without any update, I'll close this as won't fix.

closed 2022-07-22 by 0xc0170