ARMmbed/mbed-os #15167: Mbed OS network tests may not be suitable for some cellular communications

davidalo, 2021-11-17:
### Description of defect

I have been developing a new driver for a QUECTEL modem. I am trying to run the Mbed OS tests using the new porting, all of them works as expected except two:

-  `mbed-os-connectivity-netsocket-tests-tests-netsocket-udp : UDPSOCKET_ECHOTEST_BURST`                      
- `mbed-os-connectivity-netsocket-tests-tests-netsocket-udp : UDPSOCKET_ECHOTEST_BURST_NONBLOCK`

I have been reviewing those tests and they perform the following actions:
1. They send multiple UDP packets of different data and lengths to an echo server.
2. UDP packets are received using a buffer of the maximum packet length.
3. UDP packets are checked to be received (even in a different order)

The problem is that, this tests are assuming that the UDP socket API will pop a UDP packet (and no more). For example, if we receive three packets of 100, 300 and 200 bytes and we try to read 600 bytes, we will have to perform three read operations and read 100, 300 and 200 bytes respectively.

The problem is that, in some modems, such as in QUECTEL ones, UDP datagrams are stored as if they were a TCP stream, it does not have delimiters for the different received packets so when you try to read 600 bytes, it will return you the three packets (in the order they have been received).

I actually don't know how to approach this in order to make the component pass the tests.

#### Target(s) affected by this defect ?

Cellular components.

#### Toolchain(s) (name and version) displaying this defect ?

GCC_ARM 

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.14.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ? 

Running Mbed OS connectivity tests:

```bash
mbed test -n mbed-os-connectivity-netsocket*
```

mbedmain, 2021-11-17:
@davidalo thank you for raising this issue.Please take a look at the following comments:

It would help if you could also specify the versions of any tools you are using?
 
NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

davidalo, 2021-11-17:
Updated.

davidalo, 2021-11-24:
Added some extra details.

0xc0170, 2021-12-23:
cc @ARMmbed/mbed-os-connectivity