ARMmbed/mbed-os #15170: Export to uvision failing with missing context fault handler

Eulenbaer, 2021-11-18:
### Description of defect
When exporting a uVision6 project for ARMC6 toolchain on NUCLEO_F746ZG target, the build fails with the following error.
```
linking...
.\BUILD\mbed-os-example-blinky.axf: Warning: L3912W: Option 'legacyalign' is deprecated.
.\BUILD\mbed-os-example-blinky.axf: Error: L6218E: Undefined symbol mbed_fault_context (referred from .\build\except.o).
```
This error was not encountered when exporting for uVision5, ARMC5 toolchain.

#### Target(s) affected by this defect ?
Only tried to export to NUCLEO_F746ZG target, but probably others as well.

#### Toolchain(s) (name and version) displaying this defect ?
uVision6 / ARMClang6

#### What version of Mbed-os are you using (tag or sha) ?
tag: mbed-os-6.15.0 
sha: 4cfbea43cabe86bc3ed7a5287cd464be7a218938

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli

#### How is this defect reproduced ? 
1) Create a new project or take an existing one (e.g. blinky).
2) Export the project for uVision6 by using
    ````
    mbed export -i uvision6 -m NUCLEO_F746ZG --source .
    ````
3) Build the project

0xc0170, 2021-11-18:
I am closing this as won't fix. Exporters were deprecated and are no longer maintained.

I've seen the error. the problem might be with not defining `MBED_CONF_PLATFORM_CRASH_CAPTURE_ENABLED` ? If its not defined the symbol is undefined.

closed 2021-11-18 by 0xc0170

jeromecoutant, 2021-11-18:
I can reproduce the issue, and I don't have any more patch except to remove the crash feature.

In Keil uvision, open mbed_config.h file and upate:
#define MBED_CONF_PLATFORM_CRASH_CAPTURE_ENABLED     0

ATM-HSW, 2021-11-19:
> I am closing this as won't fix. Exporters were deprecated and are no longer maintained.
> 

Is this somewhere documented, that exporters are depricated? I'm reading this the first time and tried to find the deprecation warning but were not successful.

It would be crucial to know this.

ATM-HSW, 2021-11-19:
> I can reproduce the issue, and I don't have any more patch except to remove the crash feature.
> 
> In Keil uvision, open mbed_config.h file and upate: #define MBED_CONF_PLATFORM_CRASH_CAPTURE_ENABLED 0

Changing the mbed_config.h is not neccessary, just adding 
```
"platform.crash-capture-enabled": false 
```

to mbed_app.json.

Eulenbaer, 2021-11-19:
Another option is to modify the except.S assambly to explicitly include the mbed_config.h.
The mbed_config.h is only included for C/C++ files by Keil uVision projects.

In Keil uVision this might look like this:
[image]

Please be sure to add the location where mbed_config.h might be found. In my case it is 'mbed_I2CEEPROM'.