ARMmbed/mbed-os #15172: float-abi flag configuration

16L-YT, 2021-11-21:
### Description of defect
The compiler and linker flag `float-abi` to select the float engine option seems can't be configured from application. It is fixes with `soft` for GCC and `hard` for ARM compilers in the following file: `mbed-os/tools/cmake/cores/Cortex-M4F.cmake`. Howevere, In my application, I had to select the `-mfloat-abi:hard` option and the only way was to modify the file `mbed-os/tools/cmake/cores/Cortex-M4F.cmake` and change the GCC option to `hard`. This workaround is not portable as it changes a file shipped with Mbed-os.  This issue is a result of the [discussion went in the forum](https://forums.mbed.com/t/mbed-tools-uses-different-compiler-float-abi-flag-than-mbed-studio-build/15266). 

#### Target(s) affected by this defect ?
Issue does not seem to be target specific. I tried with an Ambiq target.

#### Toolchain(s) (name and version) displaying this defect ?
gcc-arm-none-eabi-9-2020-q2-update  

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.14.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-tools 7.31.1, Mbed Studio 1.4.3

#### How is this defect reproduced ? 
Try to build using `mbed-tools` any example/demo code with library compiled with `-mfloat-abi:hard`.

0xc0170, 2021-11-22:
There was related discussion some time ago https://github.com/ARMmbed/mbed-os/issues/9153

We inherited the default settings from the old tools to be consistent. They are hardcoded in core cmake files.

These core files getting in via mbed toolchain file in app.cmake

```
# Load toolchain file
if(NOT CMAKE_TOOLCHAIN_FILE OR MBED_TOOLCHAIN_FILE_USED)
    set(MBED_TOOLCHAIN_FILE_USED TRUE CACHE INTERNAL "")
    # We want to bring CMP0123 we set in mbed_toolchain.cmake
    # to the whole Mbed OS.
    include(mbed_toolchain NO_POLICY_SCOPE)
endif()
```

You could write your own toolchain file and just use CMAKE_TOOLCHAIN_FILE set to it, mbed one should not be added.

16L-YT, 2021-11-22:
Thanks for pointing `CMAKE_TOOLCHAIN_FILE` option out. I did not know about it. This is for sure solved my issue. 

For future refer: 
This option is set using similar command in the main CMakeLists.txt `set(CMAKE_TOOLCHAIN_FILE "${CMAKE_CURRENT_SOURCE_DIR}/file_name.cmake" CACHE STRING "")`. I basically copied `mbed-os/tools/cmake/cores/Cortex-M4F.cmake` content and changed the needed.

closed 2021-11-22 by 16L-YT