ARMmbed/mbed-os #15173: STM32WLE5: I2C unknown instance for I2C_1

IoTopenTech, 2021-11-21:
### Description of defect

I am building a program with two APDS9930 I2C distance sensors.

Both use the same I2C slave address, so I decided to use 2 I2C peripherals (STM32WLE5 has 3 I2C available).

I2C_2 and I2C_3 work with any of the pin configurations available.

#### Target(s) affected by this defect ?
STM32WLE5 (RAK3172)

#### Toolchain(s) (name and version) displaying this defect ?

#### What version of Mbed-os are you using (tag or sha) ?
Mbed OS 6.15

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Mbed Studio 1.4.3

#### How is this defect reproduced ? 

Trying to initialize an I2C configuration for I2C_1 using any of the available pin configurations for SDA/SCL (PA_10/PA_9, PA_10/PB_6, PB_7/PA_9, or PB_7/PB_6) always fails.

[image]

mbedmain, 2021-11-21:
@IoTopenTech thank you for raising this issue.Please take a look at the following comments:

What toolchain(s) are you using?

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

jeromecoutant, 2021-11-22:
Hi
I think this is corrected by #15133 ?

jeromecoutant, 2021-11-22:
> #### Target(s) affected by this defect ?
> STM32WLE5 (RAK3172)

FYI: https://github.com/ARMmbed/stm32customtargets#rak-wireless-rak3172

IoTopenTech, 2021-11-22:
Thank you very much @jeromecoutant 
Yes that modification solves my problem.
Kind regards from Madrid.

0xc0170, 2021-11-22:
Closing as resolved.

closed 2021-11-22 by 0xc0170