ARMmbed/mbed-os #15179: How to restrict authentication in wisun

mcuxmx, 2021-11-26:
### Description of defect

I have two wisun BR (BR1, BR2) and several routers (A1,A2... An, B1,B2...BN). Their default network parameters are the same (e.g. network name, certificate, etc.)

sometimes Ax is connected to BR2 and Bx is connected to BR1， I want limit Ax connect to BR1 and Bx to BR2, but don't want to modify the parameters of each router.
Can I change the parameters of BR , or limit it in radius server? How to do it?

#### Target(s) affected by this defect ?
N/A

#### Toolchain(s) (name and version) displaying this defect ?
N/A

#### What version of Mbed-os are you using (tag or sha) ?

N/A

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
N/A

#### How is this defect reproduced ? 
N/A

0xc0170, 2021-11-29:
@artokin would you be able to help?

artokin, 2021-11-29:
@mikter , would you please help?

mikter, 2021-12-01:
There are multiple ways to balance devices to the different networks.

Different configuration based options
* Configuring the network name to different devices and border routers. This is the real use balancing you should use in this case and in all cases in real world.
* Configuring different certificates to different network devices and then configuring border routers to reject other network devices. This causes extra authentication attempts and slows down the network and is not well tested

Internal Wi-SUN features
* Built in balancing is network size dependent, but devices selected are random and the granularity starts from 64 devices to one network after witch they start preferring the smaller network. this is in your case where all have same configuration

Pelion features 
* Scanning and configuring preferred network name/panID combinations from Pelion service. This is the real customer requirement. NOT done in anywhere not in Pelion or not in stack or applications.

Then there is this one where everything is done outside of Wi-SUN
* Hacking or doing something to Radius server so that it would prevent authentications from certain border router/device authentications. Don't know if this is in any way doable. would result lots of re-authentications and slows the network

So I suggest you change the network name for the devices and border routers or trust in randomness.

mcuxmx, 2021-12-02:
@mikter, Thanks  for your reply

closed 2021-12-02 by mcuxmx