ARMmbed/mbed-os #15182: BLE: Notification and indications are not handled when security is disabled

boraozgen, 2021-11-29:
### Description of defect

The following preprocessor switch prevents handling of notifications and indications (pushing to the client) when the security feature is disabled. Removing the preprocessor switch fixes the issue. Any idea why this code snippet has anything to do with the security switch?

https://github.com/ARMmbed/mbed-os/blob/2eb06e76208588afc6cb7580a8dd64c5429a10ce/connectivity/FEATURE_BLE/source/cordio/source/GattServerImpl.cpp#L827-L848

#### Target(s) affected by this defect ?
BLE targets.

#### Toolchain(s) (name and version) displaying this defect ?
GCC ARM

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.15.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ? 

Get the update GATT server example and disable security with: `"ble.ble-feature-security": false`. Observe no notifications being received by the client.

mbedmain, 2021-11-29:
@boraozgen thank you for raising this issue.Please take a look at the following comments:

It would help if you could also specify the versions of any tools you are using?
 
NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

0xc0170, 2021-11-29:
@ARMmbed/mbed-os-connectivity could you review?